kiddin9/Kwrt #1380: FriendlyARM NanoPC T6 无法启动

Diciya, 2023-05-26:
使用网页构建的FriendlyARM NanoPC T6刷入emmc无法启动
1，使用官方系统的emmc刷机助手 刷你的固件不能进入系统
2，使用sd系统命令刷入emmc后也无法启动
感谢大神修复，官方网盘最新源码包有更新，不知能否有助于修复

JobberRT, 2023-06-09:
@Diciya Lean has release his lede for RK3588 chip (nanopc T6 also supported)
https://github.com/coolsnowwolf/lede/releases/tag/20230609

Right now it's not open-source, just a pre-compile firmware, but I think there should be no problem

Diciya, 2023-06-09:
> @Diciya Lean has release his lede for RK3588 chip (nanopc T6 also supported) https://github.com/coolsnowwolf/lede/releases/tag/20230609
> 
> Right now it's not open-source, just a pre-compile firmware, but I think there should be no problem

Thank you, this method has solved my problem 😊