kiddin9/Kwrt #1393: x86 网页后台升级后 05.28.2023 所有页面都 Not Found

learn2s, 2023-06-01:
升级后网页后台无法访问，都显示 Not Found，ssh 能登录进去

请教有办法通过 ssh 进行版本升降级吗？或者有什么办法修复访问问题？

kiddin9, 2023-06-01:
清除下浏览器缓存 或者用完整的后台路径 http://10.0.0.1/cgi-bin/luci/ 访问

learn2s, 2023-06-01:
谢谢大佬回应，有办法通过 shell 保留配置强制刷进指定版本固件吗？我查 issues 类似问题试着改了不少东西，估计都改乱了，以上操作现在也无法访问了。提示 403 Forbidden 

> 清除下浏览器缓存 或者用完整的后台路径 http://10.0.0.1/cgi-bin/luci/ 访问

kiddin9, 2023-06-01:
上传固件到 /tmp 目录 
通过命令 sysupgrade /tmp/固件名 刷入即可.

learn2s, 2023-06-01:
sysupgrade /tmp/固件名  20230527 或 28 的包就还是一样 Not Found ，2022 10 和 08 的包就报
```
# nginx -T -c '/etc/nginx/uci.conf'
2023/06/01 11:20:12 [emerg] 8915#0: dlopen() "/usr/lib/nginx/modules/ngx_http_ubus_module.so" failed (Error loading shared library /usr/lib/nginx/modules/ngx_http_ubus_module.so: No such file or directory) in /etc/nginx/module.d/luci.module:1
nginx: configuration file /etc/nginx/uci.conf test failed

```

> 上传固件到 /tmp 目录 通过命令 sysupgrade /tmp/固件名 刷入即可.

learn2s, 2023-06-01:
包括虚拟机跑的一样，求问应该哪方面着手，不重建或者拿实体物理机重刷的话

DCLangX, 2023-06-07:
同样的现象，后来发现是nginx的配置问题，我忘了改的哪里，你打开/etc/nginx/conf.d跟我的对比下
```
location /cgi-bin/luci {
		index  index.html;
		include uwsgi_params;
		uwsgi_read_timeout 300s;
		uwsgi_param SERVER_ADDR $server_addr;
		uwsgi_modifier1 9;
		uwsgi_pass unix:////var/run/luci-webui.socket;
}
location ~ /cgi-bin/cgi-(backup|download|upload|exec) {
		include uwsgi_params;
		uwsgi_read_timeout 90s;
		uwsgi_param SERVER_ADDR $server_addr;
		uwsgi_modifier1 9;
		uwsgi_pass unix:////var/run/luci-cgi_io.socket;
}

location /luci-static {
		error_log stderr crit;
}

# location ~* ^/(.*) {
# 		return 307 $scheme://$http_host/$1$is_args$args;
# }
location = / {
		return 301 $scheme://$http_host/cgi-bin/luci/;
}

location /ubus {
        ubus_interpreter;
        ubus_socket_path /var/run/ubus/ubus.sock;
        ubus_parallel_req 2;
}

```