kiddin9/Kwrt #1397: R68S刷入本固件后,点击Recovery按钮上电无法再次进入线刷模式

Haruhi-Suzumiya, 2023-06-02:
### 使用硬件
- R68S 4G版本,上星期从淘宝购入

### Bug现象
- 无法按照刷机教程进入线刷模式

### Bug背景
- 这台机器购入时为原厂固件,按照教程配置了Windows端的环境(Win11),确认可以进入线刷模式并成功刷入了iStoreOS(20230526版本)
- 5月31日使用作者提供的网页一键编译了固件(自带软件包全部保留,常用软件包添加了iStore,其他软件未添加)并使用线刷成功刷入
- 因固件中的iStore点击软件右下角的安装无效(不会直接开始安装而是跳转[软件包]画面),决定重新刷回iStoreOS

### Bug详细
- 按照上述线刷流程未能进入刷机模式,普通模式能正常启动,,只能通过短接MaskRom触点的方式进入
- 短接触点后重新刷入本固件,点击Recovery按钮依然失败

### 自行排查
|  (当前)固件版本  | 按住Recovery上电后开发工具侧文字提示  |
|  ----  | ----  |
| iStoreOS(20230527版本)  | "发现Loader设备"  |
| 原厂(20220701版本)  | "发现Maskrom设备"  |
| 本项目固件  | "未发现设备"  |

### 总结
1. iStoreOS和原厂固件刷入后都能使用Recovery进入线刷模式,本项目固件则不可
2. 通过短接救砖方式从本项目固件改刷为原厂固件后,使用Recovery按钮进入线刷的功能被修复
3. 修复后再次刷入本固件,Bug复现,疑似刷入本固件会导致uboot失效

1441284434, 2023-06-08:
老哥解决了没，一样刷死了  可以进MaskRom，刷固件不能启动 别的也不行

Haruhi-Suzumiya, 2023-06-09:
> 老哥解决了没，一样刷死了  可以进MaskRom，刷固件不能启动 别的也不行

我跟你这个情况不一样，我是按recovery进不了maskrom，短接能进，进去也能刷
所以我进去之后刷istoreos了