kiddin9/Kwrt #1404: 小米AX3600刷机失败，请增加编译型号 Xiaomi AX3600 (stock layout)

popalanluck, 2023-06-07:
原supes.top里的AX3600，只能刷小分区固件。用supes网站定制生成的ax3600大固件，刷入后就无法启动。

请参考 https://firmware-selector.immortalwrt.org/?version=23.05-SNAPSHOT&target=ipq807x%2Fgeneric&id=xiaomi_ax3600-stock
immortalwrt源码： https://github.com/immortalwrt/immortalwrt/tree/openwrt-23.05

增加一个编译型号:      xiaomi_ax3600-stock 。
这个才适合刷了第三方Uboot与解锁大分区的小米AX3600 ,可以刷大于50Mb的ax3600固件。
[image]
[image]

kiddin9, 2023-06-07:
按照这个教程
https://supes.top/%e7%ba%a2%e7%b1%b3ax6-%e8%a7%a3%e9%94%81ssh%e5%88%b7u-boot%e4%b8%8eopenwrt%e6%95%99%e7%a8%8b/
刷入不死uboot, 在uboot web页面中刷入openwrt固件.

popalanluck, 2023-06-07:
ax3600 已经刷了不死u-boot，并解锁了大分区，但刷入supes网站定制的固件(大于50Mb)以后,无法启动。
请增加型号Xiaomi AX3600 stock，需要按Xiaomi AX3600 (stock layout)编译生成的固件。