kiddin9/Kwrt #1409: 阿里DDNS和动态DDNS，两个插件都不能实现阿里云域名的IP更新

kson34, 2023-06-10:
版本：[06.08-GENERIC-ROOTFS.TAR.GZ]
运行环境：pve的LXC
发现问题，阿里DDNS和动态DDNS，两个插件都不能实现阿里云域名的IP更新，报错如下：
[image]
[image]

0HUA0, 2023-06-19:
用ddns-go

kson34, 2023-06-24:
> 

DDNS-GO 也是报错的，感觉是底层的问题

dengke80622, 2023-06-29:
luci-app-aliddns | 0.4.0-21 我这正常

kson34, 2023-06-29:
> luci-app-aliddns | 0.4.0-21 我这正常

确定你的环境是一样吗？
版本：[06.08-GENERIC-ROOTFS.TAR.GZ]
运行环境：pve的LXC。
试过几次了一样报错，玩这块几年了不会操作错的。

kson34, 2023-06-29:
> luci-app-aliddns | 0.4.0-21 我这正常

并且相同的配置用的是1月份的版本，一直到现在也没异常。就是[06.08-GENERIC-ROOTFS.TAR.GZ]异常