kiddin9/Kwrt #1423: IPQ6000系（兆能M2等）自定义构建固件无法启动

refined-fish, 2023-06-23:
## 预期行为
通过网页https://supes.top/ 自定义编译兆能M2的固件并通过Uboot刷入使用。
## 实际情况
如题，IPQ6000系（兆能M2等）自定义构建固件刷入后无法获取IP，无法使用，但是下载网站定期自动编译的镜像固件则可以启动没有问题。尝试将硬件型号更改为CMIOT-AX18，得到的结果仍然如此，默认镜像可以使用，自定义则不行（此处的自定义包括任何自定义，我尝试的几次无论做什么改动都无法使用）
## 使用硬件
硬改过1G内存的兆能M2硬路由（ZN M2）
## 网页编译平台信息
2023-06-08 16:51:36
分支: 23.05.0-rc1
内核: 4.4.60
## 复现步骤
1. 使用复位键进入Uboot，上传UBI格式固件更新
2. 等待重启
3. 如果上传的是自定义构建的固件，则会无法获取到IP，无法进入后台
4. 如果上传的是网站自动编译的固件，则可以获取IP，能够进入后台

gimguang, 2023-06-24:
360 v6有同样的问题

kiddin9, 2023-07-01:
已修复, 重新定制即可.

qzydustin, 2023-09-06:
> 已修复, 重新定制即可.

360V6使用非自定义固件无法启动

refined-fish, 2023-09-09:
> 已修复, 重新定制即可.

问题重新出现了，编译固件无论是Uboot刷入还是先刷入老版本再更新，都会无限重启

refined-fish, 2023-09-09:
> > 已修复, 重新定制即可.
> 
> 问题重新出现了，编译固件无论是Uboot刷入还是先刷入老版本再更新，都会无限重启

顺带一提，目前的情况是即使不定制，直接使用网页提供的自动构建固件也无法启动，会无限重启