kiddin9/Kwrt #1431: luban wifi就不正常了

zyystudio, 2023-06-29:
luban 从20230418之后的编译版本开始wifi就不正常了截至到20230629为止还没有 请问 需要哪些可以供参考解决问题？

构建日志
输出日志:

Building images for ramips - JDCloud Luban 鲁班
Packages: luci-app-argon-config luci-app-ddnsto luci-app-diskman luci-app-watchcat luci-app-webadmin luci-ssl luci-theme-bootstrap autocore base-files block-mount busybox ca-bundle curl dnsmasq-full dropbear ds-lite firewall fstools iwinfo kernel kmod-gpio-button-hotplug kmod-ipt-nat kmod-ipt-nat6 kmod-leds-gpio kmod-lib-zstd kmod-mt7915e kmod-tcp-bbr libc libgcc libustream-mbedtls logd luci-app-advanced luci-app-autoreboot luci-app-firewall luci-app-gpsysupgrade luci-app-opkg luci-app-upnp luci-app-wizard luci-base luci-compat luci-lib-fs luci-lib-ipkg mtd nano netifd odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp resolveip swconfig uboot-envtools uci uclient-fetch urandom-seed urngd wget-ssl wpad-mini zram-swap

Installing packages...
Installing libc (1.2.4-4) to root...
Installing libgcc1 (12.3.0-4) to root...
Downloading file:packages/libgcc1_12.3.0-4_mipsel_24kc.ipk
Configuring libgcc1.
Configuring libc.
Installing kernel (5.15.114-1-1) to root...
Configuring kernel.
Installing luci-app-argon-config (git-24.166.45322-bba36d4) to root...
Downloading file://www/wwwroot/op.supes.top/23.05/packages/mipsel_24kc/luci-app-argon-config_git-24.166.45322-bba36d4_all.ipk
Installing kmod-crypto-hash (5.15.114-1) to root...
Downloading file:packages/kmod-crypto-hash_5.15.114-1_mipsel_24kc.ipk
Installing kmod-crypto-null (5.15.114-1) to root...
Downloading file:packages/kmod-crypto-null_5.15.114-1_mipsel_24kc.ipk
Installing kmod-crypto-aead (5.15.114-1) to root...
Downloading file:packages/kmod-crypto-aead_5.15.114-1_mipsel_24kc.ipk
Installing kmod-crypto-manager (5.15.114-1) to root...
Downloading file:packages/kmod-crypto-manager_5.15.114-1_mipsel_24kc.ipk
Installing kmod-crypto-user (5.15.114-1) to root...
Downloading file:packages/kmod-crypto-user_5.15.114-1_mipsel_24kc.ipk
Installing kmod-crypto-authenc (5.15.114-1) to root...
Downloading file:packages/kmod-crypto-authenc_5.15.114-1_mipsel_24kc.ipk
Installing kmod-cryptodev (5.15.114+1.12-ramips-1) to root...
Downloading file:packages/kmod-cryptodev_5.15.114%2b1.12-ramips-1_mipsel_24kc.ipk
Installing libatomic1 (12.3.0-4) to root...
Downloading file:packages/libatomic1_12.3.0-4_mipsel_24kc.ipk
Installing libopenssl3 (3.0.9-13) to root...
Downloading file:packages/libopenssl3_3.0.9-13_mipsel_24kc.ipk
Installing libnghttp2-14 (1.51.0-1) to root...
Downloading file:packages/libnghttp2-14_1.51.0-1_mipsel_24kc.ipk
Installing ca-bundle (20230311-1) to root...
Downloading file:packages/ca-bundle_20230311-1_all.ipk
Installing libcurl4 (8.1.2-9) to root...
Downloading file://www/wwwroot/op.supes.top/23.05/packages/mipsel_24kc/libcurl4_8.1.2-9_mipsel_24kc.ipk
Installing curl (8.1.2-9) to root...
Downloading file://www/wwwroot/op.supes.top/23.05/packages/mipsel_24kc/curl_8.1.2-9_mipsel_24kc.ipk
Installing libubox20220927 (2022-09-27-ea560134-1) to root...
Downloading file:packages/libubox20220927_2022-09-27-ea560134-1_mipsel_24kc.ipk
Installing libjson-c5 (0.16-3) to root...
Downloading file:packages/libjson-c5_0.16-3_mipsel_24kc.ipk
Installing jsonfilter (2018-02-04-c7e938d6-1) to root...
Downloading file:packages/jsonfilter_2018-02-04-c7e938d6-1_mipsel_24kc.ipk
Installing libubus20220615 (2022-06-15-9913aa61-1) to root...
Downloading file:packages/libubus20220615_2022-06-15-9913aa61-1_mipsel_24kc.ipk
Installing libuci20130104 (2023-03-05-04d0c46c-1) to root...
Downloading file:packages/libuci20130104_2023-03-05-04d0c46c-1_mipsel_24kc.ipk
Installing libblobmsg-json20220927 (2022-09-27-ea560134-1) to root...
Downloading file:packages/libblobmsg-json20220927_2022-09-27-ea560134-1_mipsel_24kc.ipk
Installing rpcd (2023-03-14-d9788300-1) to root...
Downloading file:packages/rpcd_2023-03-14-d9788300-1_mipsel_24kc.ipk
Installing rpcd-mod-file (2023-03-14-d9788300-1) to root...
Downloading file:packages/rpcd-mod-file_2023-03-14-d9788300-1_mipsel_24kc.ipk
Installing libnl-tiny2023-04-02 (2023-04-02-11b7c5f0-1) to root...
Downloading file:packages/libnl-tiny2023-04-02_2023-04-02-11b7c5f0-1_mipsel_24kc.ipk
Installing rpcd-mod-luci (20230123-1) to root...
Downloading file:packages/rpcd-mod-luci_20230123-1_mipsel_24kc.ipk
Installing libucode20220812 (2023-04-03-51638672-1) to root...
Downloading file:packages/libucode20220812_2023-04-03-51638672-1_mipsel_24kc.ipk
Installing rpcd-mod-ucode (2023-03-14-d9788300-1) to root...
Downloading file:packages/rpcd-mod-ucode_2023-03-14-d9788300-1_mipsel_24kc.ipk
Installing cgi-io (2022-08-10-901b0f04-21) to root...
Downloading file:packages/cgi-io_2022-08-10-901b0f04-21_mipsel_24kc.ipk
[... 771 lines cut ...]
But that file is already provided by package * luci-app-opkg
* check_data_file_clashes: Package luci-i18n-opkg-zh-cn wants to install file /data/cache/c8/cache/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/lua/luci/i18n/opkg.zh-cn.lmo
But that file is already provided by package * luci-app-opkg
* opkg_install_cmd: Cannot install package luci-i18n-opkg-zh-cn.
* check_data_file_clashes: Package luci-i18n-upnp-zh-cn wants to install file /data/cache/c8/cache/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/etc/uci-defaults/luci-i18n-upnp-zh-cn
But that file is already provided by package * luci-app-upnp
* check_data_file_clashes: Package luci-i18n-upnp-zh-cn wants to install file /data/cache/c8/cache/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/lua/luci/i18n/upnp.zh-cn.lmo
But that file is already provided by package * luci-app-upnp
* opkg_install_cmd: Cannot install package luci-i18n-upnp-zh-cn.
Collected errors:
* opkg_conf_parse_file: Duplicate src declaration (openwrt_smflash https://op.supes.top/packages/small_flash). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://op.supes.top/23.05/packages/mipsel_24kc/base). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://op.supes.top/23.05/packages/mipsel_24kc/packages). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://op.supes.top/23.05/packages/mipsel_24kc/luci). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://op.supes.top/23.05/packages/mipsel_24kc/routing). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_kiddin9 https://op.supes.top/23.05/packages/mipsel_24kc/kiddin9). Skipping.
Pseudo file "dev" exists in source filesystem "/data/cache/c8/cache/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/dev".
Ignoring, exclude it (-e/-ef) to override.
5933+1 records in
5933+1 records out
3037938 bytes (3.0 MB, 2.9 MiB) copied, 0.0191411 s, 159 MB/s
17947+1 records in
17947+1 records out
9189332 bytes (9.2 MB, 8.8 MiB) copied, 0.0481854 s, 191 MB/s
padding image to 00bb0000

zyystudio, 2023-06-29:
说明下就是网络菜单下 没有 无线 这个子菜单

hbzxm, 2023-08-15:
是的，的确刷入后没有无线了