kiddin9/Kwrt #1452: AX6000刷入后无法启动

pgou, 2023-07-15:
1、AX6000已刷hanwckf大佬的不死uboot，从uboot的webui刷入openwrt-07.13.2023-mediatek-filogic-xiaomi_redmi-router-ax6000-ubootmod-squashfs-factory.bin，然后无限重启，请修复。
2、参照https://www.right.com.cn/forum/thread-8272071-1-1.html用ubootmod升级openwrt-07.13.2023-mediatek-filogic-xiaomi_redmi-router-ax6000-ubootmod-squashfs-factory.bin也是无限重启。

请明确目前supes.top“Xiaomi Redmi Router AX6000 ”到底用的是哪种分区固件？
OpenWrt stock layout对应保留官方分区刷OpenWrt，我称为stock固件；
OpenWrt layout对应的就是hanwckf大佬的不死uboot(ubi 110MB)，我称为uboot固件；
OpenWrt U-Boot layout对应的是HZFrodo大佬的不死ubootmod(ubi 122.5MB)，我称为ubootmod固件

感觉还是uboot更好用，建议增加，谢谢！

holotr, 2023-07-16:
无限重启+1 
看命名和版本应该是openwrt的ubootmod

thisfanfan, 2023-07-20:
之前的UBOOT固件好用，建议改回UBOOT

logan3033, 2023-07-23:
> 之前的UBOOT固件好用，建议改回UBOOT

哪里还有之前的 UBOOT 固件吗？好像历史版本目录里只有 UBOOTMOD 的。

hhyuanyi, 2023-07-29:
我也是红米ax6000   我想换uboot关键我都找不到   我也不知道k大神用的谁的uboot   反正就是刷机就是启动不了