kiddin9/Kwrt #1453: 从你们开始不好好编译老外的源包准备夹带私货开始,这个openwrt就不再好用了

DUWENINK, 2023-07-15:
[image]
看到这么多bak了吗,每一个能用的,都是在里边点击了一键更新,后几个包全是疯狂的干着上传流量的勾当

DCLangX, 2023-07-16:
不会吧，我三年前就开始用这个openwrt了，小bug碰到不少，但是没碰到过这种流量异常的问题诶，要不你导出下虚拟机，换别人试试

DUWENINK, 2023-07-17:
> 不会吧，我三年前就开始用这个openwrt了，小bug碰到不少，但是没碰到过这种流量异常的问题诶，要不你导出下虚拟机，换别人试试
已经换成LEDE了

Brandzhou, 2023-07-24:
这是你虚拟机的自动备份吧，或者你虚拟机装了什么东西。一直用他的固件没有流量异常的情况呀

ltpop, 2023-08-10:
升级到 23.05，07.12.2023后，发现passwall2出现异常的连接，大佬们帮忙看看这是什么
[image]

ltpop, 2023-08-10:
同时/tmp/etc/passwall2/global.json中查到一些在页面上看不到的配置
[image]

ltpop, 2023-08-10:
@DUWENINK @DCLangX @Brandzhou 不知道这是什么:(

HalcyonDaysssss, 2025-02-10:
我也怀疑了，openwrt开启upnp偶然上传就会飙到10m/s，我docker全关了，代理也关了，能关的都关了，就是一直在跑上传