kiddin9/Kwrt #1455: 斐讯K3在更新了0706的版本后Wi-Fi一直是未启用状态

m1omoe, 2023-07-15:


m1omoe, 2023-07-16:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.118 (runner@fv-az442-763) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.3.0 r23228-cd17d8df2a) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Wed Jul 5 20:27:24 2023
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Phicomm K3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Hit pending asynchronous external abort (FSR=0x00001c06) during first unmask, this is most likely caused by a firmware/bootloader bug.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  [mem 0x0000000008000000-0x000000009fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   node   0: [mem 0x0000000088000000-0x000000009fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000009fffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s13644 r8192 d23220 u45056
[    0.000000] pcpu-alloc: s13644 r8192 d23220 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130784
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 508284K/524288K available (6609K kernel code, 578K rwdata, 1500K rodata, 1024K init, 301K bss, 16004K reserved, 0K cma-reserved, 393216K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C: DT/platform modifies aux control register: 0x0a130000 -> 0x3a530000
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 16 ways, 256 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x7e530001
[    0.000002] sched_clock: 64 bits at 700MHz, resolution 1ns, wraps every 4398046511103ns
[    0.000022] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0xa17102bcf3, max_idle_ns: 440795224838 ns
[    0.000044] Switching to timer-based delay loop, resolution 1ns
[    0.000280] Calibrating delay loop (skipped), value calculated using timer frequency.. 1400.00 BogoMIPS (lpj=7000000)
[    0.000293] pid_max: default: 32768 minimum: 301
[    0.000687] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000698] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001597] CPU: Testing write buffer coherency: ok
[    0.001641] CPU0: Spectre v2: using BPIALL workaround
[    0.001869] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002465] Setting up static identity map for 0x100000 - 0x10003c
[    0.002594] rcu: Hierarchical SRCU implementation.
[    0.002873] smp: Bringing up secondary CPUs ...
[    0.003427] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.003440] CPU1: Spectre v2: using BPIALL workaround
[    0.003533] smp: Brought up 1 node, 2 CPUs
[    0.003545] SMP: Total of 2 processors activated (2800.00 BogoMIPS).
[    0.003553] CPU: WARNING: CPU(s) started in wrong/inconsistent modes (primary CPU mode 0x13)
[    0.003558] CPU: This may indicate a broken bootloader or firmware.
[    0.005524] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.005551] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.005754] pinctrl core: initialized pinctrl subsystem
[    0.006385] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.006868] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.007400] thermal_sys: Registered thermal governor 'step_wise'
[    0.007618] hw-breakpoint: found 5 (+1 reserved) breakpoint and 1 watchpoint registers.
[    0.007635] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.015703] clocksource: Switched to clocksource arm_global_timer
[    0.016467] NET: Registered PF_INET protocol family
[    0.016586] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.017010] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.017043] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.017056] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.017071] TCP bind hash table entries: 1024 (order: 1, 8192 bytes, linear)
[... 340 lines cut ...]
[   17.526803] br-lan: port 1(lan1) entered forwarding state
[   18.914827] br-lan: port 4(wlan0) entered blocking state
[   18.914855] br-lan: port 4(wlan0) entered disabled state
[   18.915102] device wlan0 entered promiscuous mode
[... cut ...]
[   74.071402] br-lan: port 5(wlan1) entered disabled state
[   74.071616] device wlan1 entered promiscuous mode
[   74.115860] device wlan1 left promiscuous mode
[   74.115973] br-lan: port 5(wlan1) entered disabled state
[   84.679599] br-lan: port 5(wlan1) entered blocking state
[   84.679619] br-lan: port 5(wlan1) entered disabled state
[   84.679833] device wlan1 entered promiscuous mode
[   84.735779] device wlan1 left promiscuous mode
[   84.735898] br-lan: port 5(wlan1) entered disabled state

qy2009, 2023-07-17:
我也是。

Generic MAC80211 802.11b/g/n
设备未激活

qy2009, 2023-07-17:
`Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.299058] pci 0001:00:00.0: BAR 8: assigned [mem 0x20000000-0x20bfffff]
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.305877] pci 0001:00:00.0: BAR 9: assigned [mem 0x20c00000-0x20cfffff 64bit pref]
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.313648] pci 0001:01:00.0: BAR 2: assigned [mem 0x20000000-0x207fffff 64bit]
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.320991] pci 0001:01:00.0: BAR 4: assigned [mem 0x20c00000-0x20cfffff 64bit pref]
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.328767] pci 0001:01:00.0: BAR 0: assigned [mem 0x20800000-0x20807fff 64bit]
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.336105] pci 0001:00:00.0: PCI bridge to [bus 01]
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.341080] pci 0001:00:00.0:   bridge window [mem 0x20000000-0x20bfffff]
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.347891] pci 0001:00:00.0:   bridge window [mem 0x20c00000-0x20cfffff 64bit pref]
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.355695] pci 0001:00:00.0: 4.000 Gb/s available PCIe bandwidth (5.0 GT/s PCIe x1 link)
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.364055] pcie_iproc_bcma bcma0:9: host bridge /axi@18000000/pcie@14000 ranges:
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.371592] pcie_iproc_bcma bcma0:9:   No bus range found for /axi@18000000/pcie@14000, using [bus 00-ff]
Thu Jul  6 04:27:34 2023 kern.warn kernel: [    2.381214] pcie_iproc_bcma bcma0:9: non-prefetchable memory resource required
Thu Jul  6 04:27:34 2023 kern.err kernel: [    2.505763] pcie_iproc_bcma bcma0:9: PHY or data link is INACTIVE!
Thu Jul  6 04:27:34 2023 kern.err kernel: [    2.511965] pcie_iproc_bcma bcma0:9: no PCIe EP device detected
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.518776] bcm47xx-wdt bcm47xx-wdt.0: BCM47xx Watchdog Timer enabled (30 seconds, Software Timer)
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.527852] bcma-host-soc 18000000.axi: bus0: Bus registered
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.534465] NET: Registered PF_INET6 protocol family
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.540710] Segment Routing with IPv6
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.544424] In-situ OAM (IOAM) with IPv6
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.548460] NET: Registered PF_PACKET protocol family
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.553588] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.566815] 8021q: 802.1Q VLAN Support v1.8
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    2.571044] Registering SWP/SWPB emulation handler
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.578669] b53-srab-switch 18007000.ethernet-switch: found switch: BCM53012, rev 0
Thu Jul  6 04:27:34 2023 kern.warn kernel: [    2.718250] b53-srab-switch 18007000.ethernet-switch: Using legacy PHYLIB callbacks. Please migrate to PHYLINK!
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.728918] b53-srab-switch 18007000.ethernet-switch lan1 (uninitialized): PHY [dsa-0.0:00] driver [Generic PHY] (irq=POLL)
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.741369] b53-srab-switch 18007000.ethernet-switch lan2 (uninitialized): PHY [dsa-0.0:01] driver [Generic PHY] (irq=POLL)
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.753807] b53-srab-switch 18007000.ethernet-switch lan3 (uninitialized): PHY [dsa-0.0:02] driver [Generic PHY] (irq=POLL)
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.766324] b53-srab-switch 18007000.ethernet-switch wan (uninitialized): PHY [dsa-0.0:03] driver [Generic PHY] (irq=POLL)
Thu Jul  6 04:27:34 2023 kern.info kernel: [    2.778178] DSA: tree 0 setup
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    2.781892] UBI: auto-attach mtd5
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    2.785231] ubi0: attaching mtd5
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.401752] ubi0: scanning is finished
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.414389] ubi0: attached mtd5 (name "ubi", size 120 MiB)
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.419947] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.426851] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.433656] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.440634] ubi0: good PEBs: 955, bad PEBs: 5, corrupted PEBs: 0
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.446663] ubi0: user volume: 2, internal volumes: 1, max. volumes count: 128
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.453901] ubi0: max/mean erase counter: 5/2, WL threshold: 4096, image sequence number: 1319716540
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.463067] ubi0: available PEBs: 0, total reserved PEBs: 955, PEBs reserved for bad PEB handling: 15
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.472367] ubi0: background thread "ubi_bgt0d" started, PID 306
Thu Jul  6 04:27:34 2023 kern.info kernel: [    3.479136] block ubiblock0_0: created from ubi0:0(rootfs)
Thu Jul  6 04:27:34 2023 kern.notice kernel: [    3.484647] ubiblock: device ubiblock0_0 (rootfs) set to be root filesystem
[... cut ...]
Tue Jul 18 03:06:55 2023 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Tue Jul 18 03:06:55 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Tue Jul 18 03:06:57 2023 daemon.warn odhcpd[3161]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Tue Jul 18 03:12:16 2023 daemon.warn odhcpd[3161]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!`

qiuming335, 2023-07-18:
斐讯K3刷了0609的进不了系统，cfe重刷了0418正常。

jiaowoHD, 2023-07-24:
诸位都一样啊，K3不能启用WiFi

justin-eng, 2023-07-28:
同样的问题，k3 wifi不能用，我一直用的二月份版本没问题

m1omoe, 2023-07-29:
> 同样的问题，k3 wifi不能用，我一直用的二月份版本没问题

能分享下二月份的固件包么

jiaowoHD, 2023-07-29:
老的版本在自定义固件最下面有个历史版本可以下载历史固件可以使用但是不能自定义，我自定义的被我删除了，我也没有我换别的固件随便用了。

发自我的iPhone