kiddin9/Kwrt #1475: 360V6 7.12固件下zerotier启动不了

QindongHu, 2023-07-28:
360V6，刷了7.12最新build的固件，zerotier也成功安装了，但就是启动不了，勾了启用但没用，一直提示未启动
[image]
重启、卸载重装zerotier都试过了，都是不能启动服务
[image]
路由器状态啥的都是正常的，其他插件装了也正常使用
又试了用zerotier-cli也是不行，报错如下
[image]

iuiiuiiiu, 2023-07-30:
我编译的版本情况也和你一样，启动项里启动zerotier路由器还卡死了。

chenghj87, 2023-08-03:
我也不行。运行有报错
[root@OpenWrt:10:40 AM ~] #  /usr/bin/zerotier-cli 
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZSt28__throw_bad_array_new_lengthv: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6insertEmRKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6substrEmm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt13runtime_errorC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5eraseEN9__gnu_cxx17__normal_iteratorIPKcS4_EES9_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt13random_device7_M_initERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx117collateIcE9transformEPKcS3_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEmc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7_S_copyEPcPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEmmPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1EOS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZSt9use_facetINSt7__cxx117collateIcEEERKT_RKSt6locale: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcPKcS7_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt14basic_ifstreamIcSt11char_traitsIcEEC1ERKNSt7__cxx1112basic_stringIcS1_SaIcEEESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEmmRKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5rfindEcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEE3strEv: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7reserveEm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8capacityEv: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_disposeEv: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEPKc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE11_M_is_localEv: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7replaceEmmPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcS5_S5_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructEmc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9push_backEc: symbol not found
[... cut ...]
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8_M_eraseEmm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13find_first_ofEPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4copyEPcmm: symbol not found

FmgX, 2023-08-07:
我也遇到了类似的情况
Mon Aug  7 17:46:01 2023 daemon.err zerotier-one[30427]: Error relocating /usr/bin/zerotier-one: _ZSt28__throw_bad_array_new_lengthv: symbol not found
Mon Aug  7 17:46:01 2023 daemon.err zerotier-one[30427]: Error relocating /usr/bin/zerotier-one: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Mon Aug  7 17:46:01 2023 daemon.err zerotier-one[30427]: Error relocating /usr/bin/zerotier-one: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
查资料看好像是依赖比较旧的原因。libstdcpp6_12.3.0-4_x86_64.ipk对应的则是libstdcpp6_7.5.0-4_aarch64_cortex-a53.ipk以及
libgcc1_12.3.0-4_x86_64.ipk对应的libgcc1_7.5.0-4_aarch64_cortex-a53.ipk。
应该问题就出现在这两个依赖上面了。不过为什么版本低呢，应该是因为4.4.60核心的问题。至于怎么解决，我也不知道了。求大佬解决一下吧 @kiddin9

FmgX, 2023-08-09:
@QindongHu @iuiiuiiiu @chenghj87 。办法来了，你们可以试试
，不知道适不适合你们[https://eqing.121318.xyz/?p=268](https://eqing.121318.xyz/?p=268)

QindongHu, 2023-08-23:
> @QindongHu @iuiiuiiiu @chenghj87 。办法来了，你们可以试试 ，不知道适不适合你们https://eqing.121318.xyz/?p=268

确实就是这个问题，SSH上去wget了这个新版的ipk并opkg install之后，重启zt，果然就OK了，多谢兄台！

MaxTan, 2023-12-19:
感谢！ RAX3000M 使用kiddin9自助编译同样出现问题，在此文章得到解决方法