kiddin9/Kwrt #1525: 新安装固件后无法安装插件

Tidus-Sun, 2023-08-26:
在PVE中全新安装，配置为旁路由模式，可以ping通互联网，但无法安装任何插件，opkg update报错：
[root@OpenWrt:12:33 PM ~] # opkg update
Downloading https://mirrors.ustc.edu.cn/openwrt/23.05/targets/x86/64/5.15.118/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/23.05/targets/x86/64/5.15.118/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/base/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/base/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/packages/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/packages/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/luci/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/luci/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/routing/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/routing/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/kiddin9/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/kiddin9/Packages.gz

Collected errors:
 * opkg_download: 下载失败 https://mirrors.ustc.edu.cn/openwrt/23.05/targets/x86/64/5.15.118/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/base/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/packages/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/luci/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/routing/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.ustc.edu.cn/openwrt/23.05/packages/x86_64/kiddin9/Packages.gz, wget returned 8.

请问是什么问题？应该如何解决？

jzh33, 2023-08-26:
先搭梯子试试。

Tidus-Sun, 2023-08-26:
> 先搭梯子试试。

搭了的，主路由可以科学上网，用wget能从github正常下载的

tceasy, 2023-09-07:
源地址前加上相应前缀 如下
src/gz openwrt_core 源地址
src/gz openwrt_base 源地址
src/gz openwrt_packages 源地址
src/gz openwrt_luci 源地址
src/gz openwrt_routing 源地址

zhumingchuang, 2023-09-16:
同样问题