kiddin9/Kwrt #1537: MGV2000定制固件无法启动

huxubo, 2023-09-09:
MGV2000以前定制的固件内核是5.15的可以正常启动，
现在定制的固件网页上显示 内核: 5.15.127，实际却为6.1的内核无法启动

chenyajunai, 2023-10-13:
最近几个月我的R4S也都是这样的情况，最近几个月的固件遇到要么无法启动，要么老是重启，重启进去十几秒又是重启，现在刷会了保留的一个以前的固件先用着，软件也不敢更新，不知道是固件问题还是软件问题

huxubo, 2023-10-14:
> 最近几个月我的R4S也都是这样的情况，最近几个月的固件遇到要么无法启动，要么老是重启，重启进去十几秒又是重启，现在刷会了保留的一个以前的固件先用着，软件也不敢更新，不知道是固件问题还是软件问题

我之前用5.15内核没出过问题，那次看到有更新就升级了下，结果就无法启动了

tracy1949, 2023-10-19:
> > 最近几个月我的R4S也都是这样的情况，最近几个月的固件遇到要么无法启动，要么老是重启，重启进去十几秒又是重启，现在刷会了保留的一个以前的固件先用着，软件也不敢更新，不知道是固件问题还是软件问题
> 
> 我之前用5.15内核没出过问题，那次看到有更新就升级了下，结果就无法启动了

楼主 能提供一下5.15内核正常启动的固件吗？现在网页定制的新内核固件 不能用了，谢谢！

huxubo, 2023-10-20:
> > > 最近几个月我的R4S也都是这样的情况，最近几个月的固件遇到要么无法启动，要么老是重启，重启进去十几秒又是重启，现在刷会了保留的一个以前的固件先用着，软件也不敢更新，不知道是固件问题还是软件问题
> > 
> > 
> > 我之前用5.15内核没出过问题，那次看到有更新就升级了下，结果就无法启动了
> 
> 楼主 能提供一下 电视盒子MGV2000 5.15内核 正常启动的固件吗？现在网页定制的新内核固件 不能用了，谢谢！

http://bailuhu.win/d/Onedrive/openwrt_amlogic_s905l2_k5.15.135_2023.10.20_10.0.0.1_root.img.gz

huxubo, 2023-10-20:
> > > 最近几个月我的R4S也都是这样的情况，最近几个月的固件遇到要么无法启动，要么老是重启，重启进去十几秒又是重启，现在刷会了保留的一个以前的固件先用着，软件也不敢更新，不知道是固件问题还是软件问题
> > 
> > 
> > 我之前用5.15内核没出过问题，那次看到有更新就升级了下，结果就无法启动了
> 
> 楼主 能提供一下 电视盒子MGV2000 5.15内核 正常启动的固件吗？现在网页定制的新内核固件 不能用了，谢谢！

https://huxubo.win/d/OpenWrt/openwrt_amlogic_s905l2_k5.15.135_2023.10.20_10.0.0.1_root.img.gz

huxubo, 2023-10-20:
> > > 最近几个月我的R4S也都是这样的情况，最近几个月的固件遇到要么无法启动，要么老是重启，重启进去十几秒又是重启，现在刷会了保留的一个以前的固件先用着，软件也不敢更新，不知道是固件问题还是软件问题
> > 
> > 
> > 我之前用5.15内核没出过问题，那次看到有更新就升级了下，结果就无法启动了
> 
> 楼主 能提供一下 电视盒子MGV2000 5.15内核 正常启动的固件吗？现在网页定制的新内核固件 不能用了，谢谢！

网页定制的固件内核不是5.15的，写入U盘后看到是6.1的内核文件，建议作者搞个内核选项

tracy1949, 2023-10-20:
> https://huxubo.win/d/OpenWrt/openwrt_amlogic_s905l2_k5.15.135_2023.10.20_10.0.0.1_root.img.gz
你之前定制 5.15内核的老版本固件保存还在吗？  6.0内核的固件写入EMMC后 拔掉U盘  电视盒子就无法启动了！

huxubo, 2023-10-20:
> > https://huxubo.win/d/OpenWrt/openwrt_amlogic_s905l2_k5.15.135_2023.10.20_10.0.0.1_root.img.gz
> 你之前定制 5.15内核的老版本固件保存还在吗？  6.0内核的固件写入EMMC后 拔掉U盘  电视盒子就无法启动了！

我这固件是openwrt.ai上定制的rootfs用amlogic-s9xxx-openwrt打包的固件，内核用的是5.15.135，经过几天的使用没发现有问题

chenyajunai, 2023-10-22:
> > > 最近几个月我的R4S也都是这样的情况，最近几个月的固件遇到要么无法启动，要么老是重启，重启进去十几秒又是重启，现在刷会了保留的一个以前的固件先用着，软件也不敢更新，不知道是固件问题还是软件问题
> > 
> > 
> > 我之前用5.15内核没出过问题，那次看到有更新就升级了下，结果就无法启动了
> 
> 楼主 能提供一下 电视盒子MGV2000 5.15内核 正常启动的固件吗？现在网页定制的新内核固件 不能用了，谢谢！
没有留其他的

huxubo, 2023-10-22:
> > > > 最近几个月我的R4S也都是这样的情况，最近几个月的固件遇到要么无法启动，要么老是重启，重启进去十几秒又是重启，现在刷会了保留的一个以前的固件先用着，软件也不敢更新，不知道是固件问题还是软件问题
> > > 
> > > 
> > > 我之前用5.15内核没出过问题，那次看到有更新就升级了下，结果就无法启动了
> > 
> > 楼主 能提供一下 电视盒子MGV2000 5.15内核 正常启动的固件吗？现在网页定制的新内核固件 不能用了，谢谢！
> 没有留其他的
> 

楼上有链接