kiddin9/Kwrt #1577: 都来报一下有哪些型号的固件现在有问题启动不了

chenyajunai, 2023-10-29:
都来报一下有哪些型号的固件现在有问题，作者最近是不上网了吗，一直没有修复，我的R4S最近几个月一直是不正常，最近的直接无法启动

agkyunromb, 2023-10-30:
ABT ASR3000 2023-10-20 08:50:55 使用上游新uboot 可以启动op，但无线不工作

ooyyee, 2023-10-31:
斐讯n1可以启动,但是当USB口插有其他U盘时无法启动,每次必须去掉,才可以启动,然后再插U盘,原先的版本没这问题.

chenyajunai, 2023-10-31:
> 斐讯n1可以启动,但是当USB口插有其他U盘时无法启动,每次必须去掉,才可以启动,然后再插U盘,原先的版本没这问题.

哦，我的启动不了，还没有拔掉U盘试过，不知道

chenzd123456, 2023-10-31:
360v6, 看起来是上游仓库换了

chenyajunai, 2023-11-01:
> 360v6, 看起来是上游仓库换了

你的固件可以启动吗

teel-writ-ros, 2023-11-01:
NanoPi R6c 获取ip后局域网内设备无法联网

freekof, 2023-11-04:
ASUS RT-AC42U这两三个月的版本都无法启动

chenyajunai, 2023-11-05:
> ASUS RT-AC42U这两三个月的版本都无法启动 当是官方的可以。
我的前几天还刷了一次官方的，也是进不去，现在只能暂时用别的

chenyajunai, 2023-11-05:
> NanoPi R6c 获取ip后局域网内设备无法联网

最近几个月固件问题很大

woyeshizuileye, 2023-11-05:
> 都来报一下有哪些型号的固件现在有问题，作者最近是不上网了吗，一直没有修复，我的R4S最近几个月一直是不正常，最近的直接无法启动

有没有一种可能，这个项目只负责编译，有问题你去找源码仓库啊

cadmael, 2023-11-05:
n1修改wifi配置 信道改auto 直接死机拔插电源不会恢复，wan口映射也进不去web，wifi速率跑不到100mbps

freekof, 2023-11-06:
> > ASUS RT-AC42U这两三个月的版本都无法启动 当是官方的可以。
> > 我的前几天还刷了一次官方的，也是进不去，现在只能暂时用别的

我用的是官方 没有加任何插件。可以进去。当是官方没有PASSWALL

teel-writ-ros, 2023-11-07:
最后用回了官方的固件，用了这里的仓库才能上网。确认是固件的问题，作者给的固件和自己编译的都不行，都上不了网。

twtytei, 2023-11-07:
小米路由3刷了在线定制的最新包10.15.2023-openwrt-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin，无网络

WildSaCk, 2023-11-11:
> 最后用回了官方的固件，用了这里的仓库才能上网。确认是固件的问题，作者给的固件和自己编译的都不行，都上不了网。

换了仓库有些插件用不了吧

chenzd123456, 2023-11-20:
> > 360v6, 看起来是上游仓库换了
> 
> 你的固件可以启动吗

不能