kiddin9/Kwrt #1583: 现在download.openwrt.ai只有23.05的软件包了，请问之前的版本怎么下包呢？

wuyifan66, 2023-11-10:
我是1月的版本
[Powered by LuCI Master (git-d924631)](https://github.com/openwrt/luci) / [ArgonTheme vgit-24.016.56229-db5dda2](https://github.com/jerrykuku/luci-theme-argon)

默认配置的地址如下：
src/gz openwrt_core https://op.supes.top/targets/x86/64/5.15.86
src/gz openwrt_base https://op.supes.top/packages/x86_64/base
src/gz openwrt_packages https://op.supes.top/packages/x86_64/packages
src/gz openwrt_luci https://op.supes.top/packages/x86_64/luci
src/gz openwrt_routing https://op.supes.top/packages/x86_64/routing
src/gz openwrt_kiddin9 https://op.supes.top/packages/x86_64/kiddin9

我可以把上述地址前缀直接替换为新的吗？：
https://dl.openwrt.ai/23.05/packages/x86_64

tudoubocai, 2023-11-15:
好像是不可以，我试过，好多版本都不对应，还整崩溃过一次，重新编译新的吧。这作者早就是更新了就删旧版。如果不是逼不得已，建议弃坑，换别的作者的

anschein, 2024-02-16:
同样的问题，不给旧版的活路啊

swiftwind0405, 2024-11-15:
故意让人再掏钱