kiddin9/Kwrt #1600: K3路由器5G信号很差，提供一个其它固件提取的驱动文件，信号更好。

mingming-cn, 2023-12-02:
K3路由器5G信号很差，提供一个其它固件提取的驱动文件，信号更好。

把压缩包内的的 brcm 文件夹替换路由器中的 /lib/firmware/brcm 目录，然后重启生效。

[brcm_k3_驱动.tar.gz](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/files/13533348/brcm_k3_.tar.gz)

recyclezap, 2023-12-17:
换了这个新生成的固件也能够正常用5G了，但是2.4G还是不行。

mingming-cn, 2023-12-18:
> 换了这个新生成的固件也能够正常用5G了，但是2.4G还是不行。

2.4 G 可以固定的个频道，我在 K3 上测试 2.4G 和 5G 都是正常的。

recyclezap, 2023-12-18:
> > 换了这个新生成的固件也能够正常用5G了，但是2.4G还是不行。
> 
> 2.4 G 可以固定的个频道，我在 K3 上测试 2.4G 和 5G 都是正常的。

2.4G 固定在哪个信道可以用啊

mingming-cn, 2023-12-21:
> > > 换了这个新生成的固件也能够正常用5G了，但是2.4G还是不行。
> > 
> > 
> > 2.4 G 可以固定的个频道，我在 K3 上测试 2.4G 和 5G 都是正常的。
> 
> 2.4G 固定在哪个信道可以用啊

可以选频道 1 试试，不行再换换其它的，换好频道重启一下。

recyclezap, 2024-01-11:
> > > > 换了这个新生成的固件也能够正常用5G了，但是2.4G还是不行。
> > > 
> > > 
> > > 2.4 G 可以固定的个频道，我在 K3 上测试 2.4G 和 5G 都是正常的。
> > 
> > 
> > 2.4G 固定在哪个信道可以用啊
> 
> 可以选频道 1 试试，不行再换换其它的，换好频道重启一下。

好了

closed 2024-03-14 by mingming-cn