kiddin9/Kwrt #1653: host tls_initialize error in lxc container in PVE

fengyang0317, 2024-02-15:
host command cannot run in the lxc container in PVE 8.0.

The host kernel version is 
```
Linux pve 6.2.16-3-pve #1 SMP PREEMPT_DYNAMIC PVE 6.2.16-3 (2023-06-17T05:58Z) x86_64 GNU/Linux
```

The error message is
```
[root@Op-ipt:20:51 ~] # host
tls.c:88:tls_initialize(): fatal error: RUNTIME_CHECK(OPENSSL_init_ssl(OPENSSL_INIT_ENGINE_ALL_BUILTIN | OPENSSL_INIT_LOAD_CONFIG, NULL) == 1) failed
Aborted
```

flyhigao, 2024-03-04:
我和你遇到一样的问题,我是在qnap上用docker来跑openwrt, 在部署ddns时,遇到错误
root@OpenWrt:~# /usr/bin/host -V
tls.c:88:tls_initialize(): fatal error: RUNTIME_CHECK(OPENSSL_init_ssl(OPENSSL_INIT_ENGINE_ALL_BUILTIN | OPENSSL_INIT_LOAD_CONFIG, NULL) == 1) failed
Aborted

以前用kvm部署的就没有问题, 请问你后来时怎么解决的?

flyhigao, 2024-03-04:
补充一下,ldd看来,出问题的host是使用
libssl.so.3 => /usr/lib/libssl.so.3

以前没有问题的host是
libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1

将没问题的host拷过来由于更多依赖冲突,并不能运行

flyhigao, 2024-03-06:
这个问题阻碍我用ddns, 如果是为了ddns运行正常,我的解决方案那就是删掉/usr/bin/host,    
这样/usr/lib/ddns/dynamic_dns_functions.sh就会调用nslookup来代替host命令得到域名ip.  ddns运行正常.

kevonlee, 2024-04-09:
> 这个问题阻碍我用ddns, 如果是为了ddns运行正常,我的解决方案那就是删掉/usr/bin/host, 这样/usr/lib/ddns/dynamic_dns_functions.sh就会调用nslookup来代替host命令得到域名ip. ddns运行正常.
```
mv /etc/ssl/openssl.cnf /etc/ssl/openssl.cnf.bak
其实删了也一样，会自动生成一个

Felix-Bao, 2024-06-24:
> > 这个问题阻碍我用ddns, 如果是为了ddns运行正常,我的解决方案那就是删掉/usr/bin/host, 这样/usr/lib/ddns/dynamic_dns_functions.sh就会调用nslookup来代替host命令得到域名ip. ddns运行正常.
> 
> ```
> mv /etc/ssl/openssl.cnf /etc/ssl/openssl.cnf.bak
> ```
> 
> 其实删了也一样，会自动生成一个

我也是这个问题，试了一下确实可以，但不知道会不会引起其他问题。同时我编译时选的时Nginx，它也一直报错。我看这里的讨论，说是SSL的问题 https://github.com/immortalwrt/immortalwrt/issues/1124