kiddin9/Kwrt #1671: ZN M2最近的几个版本都有问题刷完开不了机

DifferentSparkler, 2024-03-06:
OpenWrt 11.19.2023 by Kiddin' / LuCI Master git-24.322.44728-ebf6bee
这个版本以后的刷机都开不起来

refined-fish, 2024-03-29:
> OpenWrt 11.19.2023 by Kiddin' / LuCI Master git-24.322.44728-ebf6bee 这个版本以后的刷机都开不起来

那个版本的内核是多少的？之前就有这个问题，由于ipq6000一直没有足够的信息，导致这个芯片能使用的内核一直停留在4.4好像，一旦站长没有特别修复指定版本的话，直接跟随上游编译就是无法启动的

DifferentSparkler, 2024-03-29:
11.29那个好像显示的也是6.1的内核

DifferentSparkler, 2024-03-30:
> > OpenWrt 11.19.2023 by Kiddin' / LuCI Master git-24.322.44728-ebf6bee 这个版本以后的刷机都开不起来
> 
> 那个版本的内核是多少的？之前就有这个问题，由于ipq6000一直没有足够的信息，导致这个芯片能使用的内核一直停留在4.4好像，一旦站长没有特别修复指定版本的话，直接跟随上游编译就是无法启动的

[image]
[image]
最新的版本今天在UBOOT模式下刷好了但是没有无线，就退回了11.19版本

lfsakura0, 2024-03-30:
我的一直都能开机，改了1G内存和usb的，但usb一直识别不了U盘和硬盘，我用4.4和其它源码的6.1和6.6都能用usb

DifferentSparkler, 2024-03-31:
> 我的一直都能开机，改了1G内存和usb的，但usb一直识别不了U盘和硬盘，我用4.4和其它源码的6.1和6.6都能用usb

其它固件能不能分享一下，我的也改了内存，其它好用的带无线的固件推荐下。

lfsakura0, 2024-03-31:
> > 我的一直都能开机，改了1G内存和usb的，但usb一直识别不了U盘和硬盘，我用4.4和其它源码的6.1和6.6都能用usb
> 
> 其它固件能不能分享一下，我的也改了内存，其它好用的带无线的固件推荐下。

https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=8344919&highlight=ax5
恩山论坛可以找到他的群，但znm2在6.6内核只能跑1w，6.1后面更新的我没试，usb和nss都没问题

lfsakura0, 2024-04-07:
> > > > 我的一直都能开机，改了1G内存和usb的，但usb一直识别不了U盘和硬盘，我用4.4和其它源码的6.1和6.6都能用usb
> > > 
> > > 
> > > 其它固件能不能分享一下，我的也改了内存，其它好用的带无线的固件推荐下。
> > 
> > 
> > https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=8344919&highlight=ax5 恩山论坛可以找到他的群，但znm2在6.6内核只能跑1w，6.1后面更新的我没试，usb和nss都没问题
> 
> 试了一下这个作者的，作者已经弃更6.1，只有6.6还在更新了，不知道怎么搞的，我自己编译的话连open-ssl都编译不进去，也是麻了，感觉ipq6000实在是难以折腾

6.1对于其他机型很稳定了，所以作者弃更了，而且官方主线也开始支持6.6了，6.6解决了跑分问题和cpueq的问题，也增加了一些emmc机型，编译的话我倒是没出现过啥问题，只有昨天作者更新主线的patch没改有问题，但作者也回退了，改了一下就好了，作者更新还是很频繁的，也出了docker版的固件，基本上有意义的需求和问题都会解决

DifferentSparkler, 2024-04-07:
> 6.1对于其他机型很稳定了，所以作者弃更了，而且官方主线也开始支持6.6了，6.6解决了跑分问题和cpueq的问题，也增加了一些emmc机型，编译的话我倒是没出现过啥问题，只有昨天作者更新主线的patch没改有问题，但作者也回退了，改了一下就好了，作者更新还是很频繁的，也出了docker版的固件，基本上有意义的需求和问题都会解决

一些常用的插件到哪里下载呀  能不能给个地址或者分享一下好用的源地址喜欢用bypass
pw我在一直没有调通  一样的节点bypass点几下就跑起来了

lfsakura0, 2024-04-07:
> 一些常用的插件到哪里下载呀 能不能给个地址或者分享一下好用的源地址喜欢用bypass pw我在一直没有调通 一样的节点bypass点几下就跑起来了

就kiddin9的软件库或者kenzok8的软件库，就这俩个常用的

DifferentSparkler, 2024-04-07:
> kenzok8的软件库

kenzok8的软件库地址能不能发一个

lfsakura0, 2024-04-07:
> > kenzok8的软件库
> 
> kenzok8的软件库地址能不能发一个

你GitHub里直接搜openwrt就会有他的包，旁边介绍就有库的链接