kiddin9/Kwrt #1703: 03.30.2024  未加载 nftables 规则集

lzwdavid, 2024-04-03:
R2S老版本硬件
固件版本 | OpenWrt 03.30.2024 by Kiddin' / LuCI Master git-25.088.77044-ba9f1f7
内核版本 | 6.1.82
自定义使用Firewall4(nftables)

已经安装 firewall4 ，在状态--防火墙里 显示: 未加载NFTtable规则
[image]

点击 打开iptables规则，显示：
[image]

之前也是刷小鸡固件，状态-防火墙 nftable规则集，刷的半年之前的固件版本显示没问题。
求问如何解决？

9thChasingWindGirl, 2024-04-03:
Dnsmasq编译的时候集成依赖的问题，也不算什么问题，不影响使用，那如果你要换成其他版本的Dnsmasq可能会导致启动不了，

lzwdavid, 2024-04-03:
也就是说 问题可能在 小鸡固件的出场软件包 dnsmasq-full 里了？谢谢楼上