kiddin9/Kwrt #1714: 听山上的人说，该分支在线编译留有后门和监控程序

xibeilang0616, 2024-04-11:
作为小白听了忐忑有点不放心，个人信息安全是天大的事，请问保护信息安全方面有哪些举措?

yanyuq, 2024-04-15:
求实锤，及溯源💫

long45343, 2024-04-18:
有源码可以看的吧

yanyuq, 2024-04-18:
> 有源码可以看的吧

固件源码大概率没啥问题，关键点应该在插件。一些插件可能有偷跑流量的情况。
不放心就自己编译，直接用作者这个库 fork 过来自己github在线编译还是要修改一些东西的。

mia0x75, 2024-04-23:
用K2P在线编译的版本，网络中有几十K每秒的广播，不知道怎么找原因。

xibeilang0616, 2024-04-23:
> 用K2P在线编译的版本，网络中有几十K每秒的广播，不知道怎么找原因。

可能是软件的给境外服务器传输的间谍行为

mia0x75, 2024-04-23:
> > 用K2P在线编译的版本，网络中有几十K每秒的广播，不知道怎么找原因。
> 
> 可能是软件的给境外服务器传输的间谍行为

先不要下结论，慢慢找原因。

gxhunter1982, 2024-04-23:
“………………个人信息安全是天大的事！………………作为……………………通讯枢纽中心，……………………一览无余，有无收集出卖转让……………………敏感信息? 作为灰色收入和不法收入来源，………………有无给西方境外政府、军事、间谍组织………………提供………………，这些广大用户被动一无所知，………………保护信息安全方面………………举措? 建立起和广大用户之间长期的信任关系”

把主要内容浓缩了下
本想说点啥，发现…………
哈哈哈哈哈哈哈哈哈哈
还以为在看体制内的报告呢

lfsakura0, 2024-04-23:
要笑死我吗，“个人信息安全”已经绷不住了，这么注意个人信息安全那平时接的无用电话和短信肯定没有哦，建议自己开发一个系统最安全，别人的多少有问题，最好去深山老林里，不上网最安全。

Aethersailor, 2024-04-25:
屁民上个网还上升到国家安全层面了，你配吗
话说回来，我的敏感信息我宁愿给“西方国家”拿去也不敢用国内设备

qqqiuuu, 2024-06-18:
作为小白，怕你就别用啊，项目都是开源的，你可以自己审查代码，再不济路由流量的去向也可以跟踪。还是那句话，不懂还怕就别用呢

wgjwillis, 2024-07-01:
证据、证据、证据，别道听途说。如果，没能力、也不想去学习“怎样找到后门”，就随便用。要真正的安全：别用计算机、手机，等可以互联的设备。国内流行的免费app，从tb到vx，哪个敢把源代码放到github上让大家审核？

wenkinliao, 2025-03-13:
确实偷流量，用了这个固件，我上行流量一下子夸张到不行，差点被电信封网。

[image]

wenkinliao, 2025-03-14:
> 确实偷流量，用了这个固件，我上行流量一下子夸张到不行，差点被电信封网。
> 
> [image]
事后查，发现是iKuai的繁星在大量使用上行流量。现在一直监控着kwrt的行为，目前好像没发现什么异常的大流量上行。再持续观查一段时间看。