kiddin9/Kwrt #1736: 小米ax3000t的固件版本2024-03-30又开不了机了

masx200, 2024-04-30:
小米ax3000t的固件版本2024-03-30又开不了机了

masx200, 2024-04-30:
是不是只有`openwrt-02.01.2024-mediatek-filogic-xiaomi_mi-router-ax3000t-squashfs-sysupgrade`版本才能开机啊?

masx200, 2024-04-30:
关于固件
型号:	Xiaomi Mi Router AX3000T
平台:	mediatek/filogic
版本:	分支: 23.05.3
内核: 6.1.82
生成日期:	2024-03-30 05:29:01
说明:	后台: 10.0.0.1  密码: root

masx200, 2024-04-30:
有没有人试过啊?是我特殊个例吗?

long45343, 2024-05-02:
现在这个05-01的可以吗
03-30的我也开不了机

yangmengli, 2024-12-03:
2024年1125也开不了机，开机以后一直重启，在ssh里刷这个https://dl.openwrt.ai/releases/24.10/targets/mediatek/filogic/kwrt-11.25.2024-mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-factory.ubi可以启动，然后在刷https://dl.openwrt.ai/releases/24.10/targets/mediatek/filogic/kwrt-11.25.2024-mediatek-filogic-xiaomi_mi-router-ax3000t-squashfs-sysupgrade.bin时就一直重启