kiddin9/Kwrt #1739: upnp问题

lfsakura0, 2024-05-02:
问题：log:miniupnpd[465]: private/reserved address x.x.x.x is not suitable for external IP

找到的原因和解决方案：
以下引用自恩山，新版miniupnpd（upnp后端）多了一个IPv4公网地址检测环节，需要WAN口地址为公网地址时，miniupnpd才会工作。 如果WAN不是公网地址，miniupnpd会在 [管理后台-状态-系统日志] 中有提示[miniupnpd: **** is not suitable for external IP]
istoreos的作者用这条commit解决这个问题https://github.com/jjm2473/luci/commit/122b947b5a6ebd8105ef405acb078cd2306e98f6

图片：
[image]
[image]

能否有空修一下这个

qwazx, 2024-05-06:
我也到了不能用的问题，我还看到有和mwan3冲突，但我没装mwan3，应该就是UPNP的问题

Benjamin1919, 2024-05-09:
试试下面3种办法：
1. 想办法让WAN口获取公网IP；
2. UPNP高级设置：勾选“使用STUN”，STUN主机填stun.miwifi.com或stun.chat.bilibili.com，STUN端口填3478；
3. 安装libupnp（这个应该没什么用）

这是miniupnp插件本身的问题，不是固件的问题，其他人开发的固件也会遇到这个问题

nevergone2020, 2024-05-13:
我这从低版本一路升上来的，upnp一直正常，现在是2024.02.01，望参考
目标平台 | x86/64
内核版本 | 5.15.148
固件版本 | OpenWrt 02.01.2024 by Kiddin' / LuCI Master git-25.050.75038-257d60d
luci-app-upnp | git-25.050.75038-257d60d

[image]

nothesong, 2024-06-04:
我是安装旧版的miniupnpd才可以用，可以参考一下
https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/issues/1745#issuecomment-2147503881

xiankaiyao, 2024-06-27:
> 我是安装旧版的miniupnpd才可以用，可以参考一下 [#1745 (comment)](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/issues/1745#issuecomment-2147503881)

请问怎么安装旧版的插件，从软件包里安装只有最新的啊