kiddin9/Kwrt #1743: 05.02版旁路由模式无法正常运行（NANOPI R2S）

Albert9364, 2024-05-04:
系统原始设置下能正常运行，“网络--接口--LAN”中手动设置旁路由配置（主要是把10.0.0.1改成192.168.X.X）后，点保存并应用无效；在“网络向导”或“系统--设置向导”中可以修改设置旁路由配置，但生效大概3分钟后，系统就无法正常运行，连后台界面都无法顺利进入。

型号：FriendlyARM NanoPi R2S / R2S Plus
平台：rockchip/armv8
出现问题的版本：分支: 23.05-SNAPSHOT；内核: 6.6.29
05.02-FRIENDLYARM_NANOPI-R2S-SQUASHFS-SYSUPGRADE.IMG.GZ

现已刷回OpenWrt 03.30.2024 by Kiddin'版本

Albert9364, 2024-05-20:
版本:	分支: 23.05-SNAPSHOT
内核: 6.6.30
生成日期:	2024-05-14 04:08:34

已解决，目前运行时间5d 23h 47m 44s

closed 2024-05-20 by Albert9364

master-yun-yun, 2024-05-25:
> 版本: 分支: 23.05-SNAPSHOT 内核: 6.6.30 生成日期: 2024-05-14 04:08:34
> 
> 已解决，目前运行时间5d 23h 47m 44s

咋解决的？
舍不得分享？

Albert9364, 2024-05-25:
> > 版本: 分支: 23.05-SNAPSHOT 内核: 6.6.30 生成日期: 2024-05-14 04:08:34
> > 已解决，目前运行时间5d 23h 47m 44s
> 
> 咋解决的？ 舍不得分享？

不是写了2024-05-14 04:08:34的 23.05-SNAPSHOT、内核6.6.30分支吗，升级、安装到这版能正常运行不就行了。程序不是我写的，系统不是我编译的，我哪知道咋解决的。