kiddin9/Kwrt #1745: upnp无法正常使用

qwazx, 2024-05-06:
无法正常进行端口转发，修改过设置，比如启用 IGDv1 模式，删除可能冲突的MWAN3插件，但均无法正常使用，插件正常启动，但就是没有出现端口映射。
[image]

ltpop, 2024-05-18:
这个可能只是luci显示的问题，可以查看iptables输出是否有内容，如果有对应内网IP的记录，就是生效了
`iptables -nvL MINIUPNPD`

qwazx, 2024-05-18:
> 这个可能只是luci显示的问题，可以查看iptables输出是否有内容，如果有对应内网IP的记录，就是生效了 `iptables -nvL MINIUPNPD`

感谢，我测试一下，我感觉好像也确实生效了但就是没显示

nothesong, 2024-06-04:
[miniupnpd_2.0.20170421-4_aarch64_cortex-a53.zip](https://github.com/user-attachments/files/15551845/miniupnpd_2.0.20170421-4_aarch64_cortex-a53.zip)
从op镜像源找到旧版的upnp安装就可以使用了，检测全部通过
[image]