kiddin9/Kwrt #1747: 京东云百里固件无法开机

Mei-Sakura, 2024-05-07:
机型：JDCloud AX6000 Baili(RE-CP-03)
问题：京东云百里使用单分区uboot刷入固件无法开机
uboot来源：https://drive.wrt.moe/uboot/mediatek/

刷入hanwckf和237源码编译固件可以正常开机

[image]

kykdu, 2024-05-12:
基于官方 OpenWrt 的固件，必须刷入官方的 U-Boot，不兼容基于 bl-mt798x 项目的 U-Boot。参考资料：https://www.right.com.cn/forum/thread-8328967-1-1.html

itq5, 2024-06-05:
官方的 U-Boot可以哪里下载？

itq5, 2024-06-05:
> 机型：JDCloud AX6000 Baili(RE-CP-03) 问题：京东云百里使用单分区uboot刷入固件无法开机 uboot来源：https://drive.wrt.moe/uboot/mediatek/
> 
> 刷入hanwckf和237源码编译固件可以正常开机
> 
> ![QQ截图20240507190549](https://private-user-images.githubusercontent.com/81066884/328500460-59d3546d-20e3-49ed-aca0-ccf8477293b5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1NTYwMjcsIm5iZiI6MTcxNzU1NTcyNywicGF0aCI6Ii84MTA2Njg4NC8zMjg1MDA0NjAtNTlkMzU0NmQtMjBlMy00OWVkLWFjYTAtY2NmODQ3NzI5M2I1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA1VDAyNDg0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg5ODNkNDUwNWQ0NTI2YTkyYjk0OGI1Y2YyMTM1NWM4YTQ2NjQ2MzRlNTA0NmIzYjRhYmFlN2UzNjY1ZGU0ZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VAv3YoN2KKX-rrPfPD_tenO9uzsVmr39ibEVGieYERM)

大佬，这个你搞定了吗？我也遇到这个问题了

Mei-Sakura, 2024-06-05:
> 大佬，这个你搞定了吗？我也遇到这个问题了

目前openwrt.ai的固件，包括官方主线和imm主线snap版本均无法正常使用/开机
imm的23.05.2版本可正常使用或者使用xwrt
[image]

目前百里固件只建议使用237/hanwckf源码编译的固件

itq5, 2024-06-30:
> 目前openwrt.ai的固件，包括官方主线和imm主线snap版本均无法正常使用/开机 imm的23.05.2版本可正常使用或者使用xwrt [...]

似乎官方的可以刷入，不知道这个openwrt.ai的能否刷入了，https://www.right.com.cn/forum/thread-8384741-1-1.html?btwaf=76804681

kykdu, 2024-07-04:
想刷入 itb 格式的官方固件，可参照 https://github.com/openwrt/openwrt/pull/14144 。itb格式没有图形界面，需自己安装luci，不会使用 opkg 的请不要尝试。

haroldhuanghao, 2024-12-30:
此机器有办法刷入kwrt吗