kiddin9/Kwrt #1749: 红米AC2100使用pppoe拨号上网，经常断网，附断网日志。有没有哪位大佬可以指点一下。谢谢

liubin-123, 2024-05-08:
Wed May  8 10:54:13 2024 daemon.notice netifd: Network device 'wan' link is down
Wed May  8 10:54:13 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Wed May  8 10:54:13 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Wed May  8 10:54:13 2024 kern.info kernel: [ 6639.800072] mtk_soc_eth 1e100000.ethernet wan: Link is Down
Wed May  8 10:54:13 2024 daemon.info pppd[3528]: Terminating on signal 15
Wed May  8 10:54:13 2024 daemon.info pppd[3528]: Connect time 110.0 minutes.
Wed May  8 10:54:13 2024 daemon.info pppd[3528]: Sent 25118602 bytes, received 67683667 bytes.
Wed May  8 10:54:13 2024 daemon.err miniupnpd[4492]: ioctl(s, SIOCGIFADDR, ...): Address not available
Wed May  8 10:54:13 2024 daemon.err miniupnpd[4492]: Failed to get IP for interface pppoe-wan
Wed May  8 10:54:13 2024 daemon.warn miniupnpd[4492]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Wed May  8 10:54:13 2024 daemon.err odhcp6c[4374]: Failed to send RELEASE message to ff02::1:2 (Network unreachable)
Wed May  8 10:54:13 2024 daemon.notice netifd: Interface 'wan_6' is disabled
Wed May  8 10:54:13 2024 daemon.notice netifd: Network device 'pppoe-wan' link is down
Wed May  8 10:54:13 2024 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Wed May  8 10:54:13 2024 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Wed May  8 10:54:14 2024 daemon.err odhcp6c[4374]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Wed May  8 10:54:14 2024 daemon.notice netifd: wan_6 (4374): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan_6" } (Permission denied)
Wed May  8 10:54:14 2024 daemon.notice netifd: Interface 'wan_6' is now down
Wed May  8 10:54:14 2024 daemon.notice netifd: Interface 'wan6' is now down
Wed May  8 10:54:15 2024 daemon.warn odhcpd[2193]: No default route present, overriding ra_lifetime!
Wed May  8 10:54:17 2024 daemon.notice netifd: Network device 'wan' link is up
Wed May  8 10:54:17 2024 daemon.notice netifd: Interface 'wan6' has link connectivity
Wed May  8 10:54:17 2024 daemon.notice netifd: Interface 'wan6' is setting up now
Wed May  8 10:54:17 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Wed May  8 10:54:17 2024 kern.info kernel: [ 6643.970025] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 1Gbps/Full - flow control off
Wed May  8 10:54:18 2024 daemon.notice netifd: Interface 'wan' is now down
Wed May  8 10:54:18 2024 daemon.notice netifd: Interface 'wan' is setting up now
Wed May  8 10:54:19 2024 daemon.notice netifd: wan (10651): uci: Entry not found
Wed May  8 10:54:19 2024 daemon.notice netifd: wan (10651): sh: out of range
Wed May  8 10:54:19 2024 daemon.warn odhcpd[2193]: No default route present, overriding ra_lifetime!
Wed May  8 10:54:19 2024 daemon.info pppd[10690]: Plugin pppoe.so loaded.
Wed May  8 10:54:19 2024 daemon.info pppd[10690]: PPPoE plugin from pppd 2.4.9
Wed May  8 10:54:19 2024 daemon.notice pppd[10690]: pppd 2.4.9 started by root, uid 0
Wed May  8 10:54:19 2024 daemon.notice netifd: Network device 'wan' link is down
Wed May  8 10:54:19 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Wed May  8 10:54:19 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Wed May  8 10:54:19 2024 kern.info kernel: [ 6646.039966] mtk_soc_eth 1e100000.ethernet wan: Link is Down
Wed May  8 10:54:20 2024 daemon.err pppd[10690]: select (waitForPADO): Interrupted system call
Wed May  8 10:54:20 2024 daemon.warn pppd[10690]: Timeout waiting for PADO packets
Wed May  8 10:54:20 2024 daemon.notice netifd: Interface 'wan6' is now down
Wed May  8 10:54:20 2024 daemon.err pppd[10690]: Unable to complete PPPoE Discovery
Wed May  8 10:54:20 2024 daemon.info pppd[10690]: Exit.
Wed May  8 10:54:20 2024 daemon.notice netifd: Interface 'wan' is now down
Wed May  8 10:54:21 2024 daemon.warn odhcpd[2193]: No default route present, overriding ra_lifetime!
Wed May  8 10:54:26 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.241:49404
Wed May  8 10:54:56 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.241:49404
Wed May  8 10:55:11 2024 daemon.notice netifd: Network device 'wan' link is up
Wed May  8 10:55:11 2024 daemon.notice netifd: Interface 'wan6' has link connectivity
Wed May  8 10:55:11 2024 daemon.notice netifd: Interface 'wan6' is setting up now
Wed May  8 10:55:11 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Wed May  8 10:55:11 2024 daemon.notice netifd: Interface 'wan' is setting up now
Wed May  8 10:55:11 2024 kern.info kernel: [ 6698.038471] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 1Gbps/Full - flow control off
Wed May  8 10:55:12 2024 daemon.notice netifd: wan (11237): uci: Entry not found
Wed May  8 10:55:12 2024 daemon.notice netifd: wan (11237): sh: out of range
Wed May  8 10:55:12 2024 daemon.info pppd[11254]: Plugin pppoe.so loaded.
Wed May  8 10:55:12 2024 daemon.info pppd[11254]: PPPoE plugin from pppd 2.4.9
Wed May  8 10:55:12 2024 daemon.notice pppd[11254]: pppd 2.4.9 started by root, uid 0
Wed May  8 10:55:27 2024 daemon.warn pppd[11254]: Timeout waiting for PADO packets
Wed May  8 10:55:27 2024 daemon.err pppd[11254]: Unable to complete PPPoE Discovery
Wed May  8 10:55:27 2024 daemon.info pppd[11254]: Exit.
Wed May  8 10:55:27 2024 daemon.notice netifd: Interface 'wan' is now down
Wed May  8 10:55:27 2024 daemon.notice netifd: Interface 'wan' is setting up now
Wed May  8 10:55:27 2024 daemon.notice netifd: wan (11265): uci: Entry not found
Wed May  8 10:55:27 2024 daemon.notice netifd: wan (11265): sh: out of range
Wed May  8 10:55:27 2024 daemon.info pppd[11295]: Plugin pppoe.so loaded.
Wed May  8 10:55:27 2024 daemon.info pppd[11295]: PPPoE plugin from pppd 2.4.9
Wed May  8 10:55:27 2024 daemon.notice pppd[11295]: pppd 2.4.9 started by root, uid 0
Wed May  8 10:55:28 2024 daemon.warn odhcpd[2193]: No default route present, overriding ra_lifetime!
Wed May  8 10:55:32 2024 daemon.info pppd[11295]: PPP session is 46501
Wed May  8 10:55:32 2024 daemon.warn pppd[11295]: Connected to 48:f8:db:f8:38:a1 via interface wan
[... 100 lines cut ...]
Wed May  8 11:06:38 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:38 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:38 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:38 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:38 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:38 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:38 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:41 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.105:1900
Wed May  8 11:06:56 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.241:49404
[... cut ...] 4492]: Unknown udp packet received from 192.168.10.241:49404
Wed May  8 11:07:56 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.241:49404
Wed May  8 11:08:26 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.241:49404
Wed May  8 11:08:51 2024 daemon.err uhttpd[2407]: [info] luci: accepted login on / for root from 192.168.10.1
Wed May  8 11:08:56 2024 daemon.notice miniupnpd[4492]: Unknown udp packet received from 192.168.10.241:49404

BI7PRK, 2024-05-08:
可能电源不够力

liubin-123, 2024-05-08:
> 可能电源不够力

但是用光猫拨号上网，就没有出现断网情况。