kiddin9/Kwrt #1757: 请求添加京东云 雅典娜 AX6600

taotieren, 2024-05-19:
参考 [开启 ssh](https://www.bilibili.com/read/cv16348109/) [拆解](https://www.acwifi.net/18599.html)

taotieren, 2024-06-21:
[京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双分区刷机+回原厂+USB救砖教程20240510](https://github.com/lgs2007m/Actions-OpenWrt/blob/main/%E5%88%B7%E6%9C%BA%E6%95%99%E7%A8%8B/%E4%BA%AC%E4%B8%9C%E4%BA%91AX1800-Pro%E4%BA%9A%E7%91%9F_AX6600%E9%9B%85%E5%85%B8%E5%A8%9C%E5%88%86%E5%8C%BA%E5%A4%87%E4%BB%BD%2BTTL%E5%8F%8C%E5%88%86%E5%8C%BA%E5%88%B7%E6%9C%BA%2B%E5%9B%9E%E5%8E%9F%E5%8E%82%2BUSB%E6%95%91%E7%A0%96%E6%95%99%E7%A8%8B20240510.md)

taotieren, 2024-06-21:
[OpenWrt — IPQ60XX云编译](https://github.com/phyng/OpenWrt/tree/main)

taotieren, 2024-06-23:
使用这边编译的 [JDCloud AX6600 factory.bin](https://dl.openwrt.ai/releases/targets/qualcommax/ipq60xx/openwrt-06.21.2024-qualcommax-ipq60xx-jdcloud_re-cs-02-squashfs-factory.bin) 刷机后不进系统，

dvb001, 2024-08-09:
不要用 factory, 用 recovery, factory 的 kernel 大小不匹配

> 使用这边编译的 [JDCloud AX6600 factory.bin](https://dl.openwrt.ai/releases/targets/qualcommax/ipq60xx/openwrt-06.21.2024-qualcommax-ipq60xx-jdcloud_re-cs-02-squashfs-factory.bin) 刷机后不进系统，