kiddin9/Kwrt #1759: r2s最新的构建版本lan口无法下发ip

liaoke01, 2024-05-23:
连接路由器或者电脑无法进入后台

master-yun-yun, 2024-05-25:
首次设置，难道不是手动指定ip后进入？
不过貌似DHCP是可能有问题
你看我反馈的，无法静态IP绑定
估计可能是1个问题

liaoke01, 2024-05-25:
> 首次设置，难道不是手动指定ip后进入？ 不过貌似DHCP是可能有问题 你看我反馈的，无法静态IP绑定 估计可能是1个问题

以前的版本中，直接用网线相连，一般情况下会自动分配ip，不需要手动指定，但自从升级到最新的版本以来，lan口。就无法分配ip导致无法有线桥接路由器或者连接电脑，即使可以分配到路由器，但是路由器散发的WiFi，手机连不上，或者说连上了务参数都正常，但就是上不了网。但是根据宽带人员的说法wan口。是可以正常拨号的。如果换回旧版本，比如说lede，又可以正常分配了，估计是出问题了，希望来个大佬解决，经过我多天的排错，应该不是我的问题，大概率固件问题，毕竟用旧版本固件可以使用

master-yun-yun, 2024-05-25:
> > 首次设置，难道不是手动指定ip后进入？ 不过貌似DHCP是可能有问题 你看我反馈的，无法静态IP绑定 估计可能是1个问题
> 
> 以前的版本中，直接用网线相连，一般情况下会自动分配ip，不需要手动指定，但自从升级到最新的版本以来，lan口。就无法分配ip导致无法有线桥接路由器或者连接电脑，即使可以分配到路由器，但是路由器散发的WiFi，手机连不上，或者说连上了务参数都正常，但就是上不了网。但是根据宽带人员的说法wan口。是可以正常拨号的。如果换回旧版本，比如说lede，又可以正常分配了，估计是出问题了，希望来个大佬解决，经过我多天的排错，应该不是我的问题，大概率固件问题，毕竟用旧版本固件可以使用

我也遇见：
1）主路由器是LINKSYS WRT1900ACS V2，有线正常，无线能连上，但无线连上后获取不到IP地址和路由器网关，导致无法上网，断开，重连，一两次后又正常了，就是感觉要延迟几分钟才能DHCP获取ip、网关等信息。
2）无线连接上后，显示已连接，，，但无internet连接，此时又有2种情况，一种是啥网站都上不了，一种是只能上百度，其余都不行，QQ也无法登录，应该不是DNS问题。
3）主路由下LAN口接H3C NX30 PRO的LAN口，手机连H3C NX30 PRO啥时候都正常；但另外一个华硕AC68U就有时候正常，有时候不正常，情况同1）、2）中情况。
4）有线连3）中的2个路由都正常。
5）主路由上无法绑定静态IP地址（具体见我反馈）
感觉是固件还是有些问题

liaoke01, 2024-05-26:
> 我也遇见： 1）主路由器是LINKSYS WRT1900ACS V2，有线正常，无线能连上，但无线连上后获取不到IP地址和路由器网关，导致无法上网，断开，重连，一两次后又正常了，就是感觉要延迟几分钟才能DHCP获取ip、网关等信息。 [...]

确实感觉是固件的问题，用旧版本的固件或者lede就不会出现类似的问题，我感觉应该是openwrt官方源代码有问题，应该要去那个地方再建立一个issue

liaoke01, 2024-05-27:
> 我也遇见： 1）主路由器是LINKSYS WRT1900ACS V2，有线正常，无线能连上，但无线连上后获取不到IP地址和路由器网关，导致无法上网，断开，重连，一两次后又正常了，就是感觉要延迟几分钟才能DHCP获取ip、网关等信息。 [...]

经过我的测试确实是lan的dhcp服务器出了问题，按照你的方法。我进行手动设置ip，就可以正常连接上了，但是如果路由器设置为桥接或者dhcp模式就不可以

master-yun-yun, 2024-05-29:
现在我把ap的dhcp开启，目前正常。
相当于弃用openwrt的dhcp服务。

liaoke01, 2024-05-30:
> 现在我把ap的dhcp开启，目前正常。 相当于弃用openwrt的dhcp服务。

我和你一样，我还想用有线桥接的，可惜用不，只能手动设置ip，不使用openwrt dhcp，使用路由器内置的dhcp服务器下发

kyon777, 2024-05-31:
> > 现在我把ap的dhcp开启，目前正常。 相当于弃用openwrt的dhcp服务。
> 
> 我和你一样，我还想用有线桥接的，可惜用不，只能手动设置ip，不使用openwrt dhcp，使用路由器内置的dhcp服务器下发

哪个版本是正常的？我想刷回去了，因为DHCP有问题虽然可以手动改IPV4  但是IPV6 就不分发了

liaoke01, 2024-06-13:
> > > 现在我把ap的dhcp开启，目前正常。 相当于弃用openwrt的dhcp服务。
> > 
> > 
> > 我和你一样，我还想用有线桥接的，可惜用不，只能手动设置ip，不使用openwrt dhcp，使用路由器内置的dhcp服务器下发
> 
> 哪个版本是正常的？我想刷回去了，因为DHCP有问题虽然可以手动改IPV4 但是IPV6 就不分发了

用immortal,这个dhcp是正常的.但是现在openwrt.AI已经有新的固件了，你试一下