kiddin9/Kwrt #1771: 6月7日发布版x86固件5.15.158几个问题，请关注解决。

dengzg2002, 2024-06-08:
1、电脑登录管理后台，系统即重启。
2、外网访问速度变慢。
3、passwall无法手动更新到最新版。

master-yun-yun, 2024-06-08:
dhcp还是不正常
无法分配IP地址

ZGGX, 2024-06-11:
我的硬件配置是D2550     5.15.148内核正常
从5.15.150就出现问题
表现为5.15.148更新dnsmasq-full.dns2tcp.chinadns-ng 3个插件就会出现问题 表现为无法分配IP
刷最新内核的5.15.158 电脑可获取到IP地址 但要进入登录界面非常缓慢《表现为浏览器一直在转圈》并且OPENWRT系统会重启

Greeed-islanl, 2024-06-15:
我的5.15.148换到158就不能PPPOE拨号了 应该是这个内核有问题 等内核更新吧

LunaticTian, 2024-06-17:
> 我的硬件配置是D2550 5.15.148内核正常 从5.15.150就出现问题 表现为5.15.148更新dnsmasq-full.dns2tcp.chinadns-ng 3个插件就会出现问题 表现为无法分配IP 刷最新内核的5.15.158 电脑可获取到IP地址 但要进入登录界面非常缓慢《表现为浏览器一直在转圈》并且OPENWRT系统会重启

跟你一样，最近两个月的版本都是这种情况...用https://github.com/kenzok8/openwrt_Build 是OK的，即使不加入插件也有问题....真奇怪。

dengzg2002, 2024-06-22:
手动升级passwall到4.77-6版本时，重启报如下错误：
Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: error loading module 'luci.controller.passwall' from file '/usr/lib/lua/luci/controller/passwall.lua':
/usr/lib/lua/luci/controller/passwall.lua:1: unexpected symbol

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function build_pagetree (/usr/share/ucode/luci/dispatcher.uc:382:73)
  called from function menu_json (/usr/share/ucode/luci/dispatcher.uc:439:26)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:898:24)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^