kiddin9/Kwrt #1776: 构建红米AX6000固件的时候大量错误日志，并且不死uboot刷入无限失败

yixuan-ovo, 2024-06-14:
定制软件包列表：
luci-app-accesscontrol luci-app-adguardhome luci-app-arpbind luci-app-homeassistant luci-app-mwan3 luci-app-netdata luci-app-nlbwmon luci-app-oaf luci-app-ramfree luci-app-syncdial luci-app-turboacc luci-app-wrtbwmon luci-theme-edge luci-app-lucky luci-app-sqm luci-app-easymesh luci-app-unishare luci-app-pushbot luci-app-ddnsto

错误日志:

Collected errors:
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_base': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_packages': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_luci': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_routing': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_kiddin9': Operation not permitted.
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/base/Packages
Updated list of available packages in /data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_base
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/packages/Packages
Updated list of available packages in /data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_packages
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/luci/Packages
Updated list of available packages in /data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_luci
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/routing/Packages
Updated list of available packages in /data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_routing
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9/Packages
Updated list of available packages in /data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/openwrt_kiddin9
Downloading file:packages/Packages
Updated list of available packages in /data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c6/cache/mediatek/filogic/dl/imagebuilder
Collected errors:
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/luci-app-accesscontrol_git-24.217.56735-8015371_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/luci-app-adguardhome_git-24.217.56735-8015371_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/libcurl4_8.7.1-r1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/curl_8.7.1-r1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/adguardhome_0.107.43-113_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/luci-app-arpbind_git-24.217.56735-8015371_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/libbpf1_1.2.2-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/ip-full_6.3.0-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/luci-app-ddnsto_git-24.237.45156-121e6d1_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/ddnsto_3.0.4-29_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/luci-app-easymesh_git-24.343.44748-87a8353_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/batctl-tiny_2023.1-2_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/libgpg-error_1.47-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/libgcrypt_1.10.2-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/umdns_2024-03-04-7c675979-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/dawn_2023-05-14-e036905a-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/luci-app-homeassistant_git-24.343.44748-87a8353_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/libmount1_2.39-2_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/lsblk_2.39-2_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/docker_26.1.0-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/libattr_2.5.1-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-hg6DcY/liblzo2_2.10-4_aarch64_cortex-a53.ipk': Operation not permitted.
[... cut ...]
./etc/init.d/appfilter: line 3: /lib/functions.sh: No such file or directory
Pseudo file "dev" exists in source filesystem "/data/cache/c6/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/dev".
Ignoring, exclude it (-e/-ef) to override.
102443+1 records in
102444+0 records out
104902656 bytes (105 MB, 100 MiB) copied, 0.26417 s, 397 MB/s
/data/cache/c6/cache/mediatek/filogic/staging_dir/host/bin/tar: Option --mtime: Treating date '@1718184306' as 2024-06-12 17:25:06

nhafeng, 2024-06-23:
最近构建的固件刷进红米AX6000后，无线菜单不见了。

nhafeng, 2024-06-24:
错误日志:

Collected errors:
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_base': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_packages': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_luci': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_routing': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_kiddin9': Operation not permitted.
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/base/Packages
Updated list of available packages in /data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_base
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/packages/Packages
Updated list of available packages in /data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_packages
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/luci/Packages
Updated list of available packages in /data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_luci
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/routing/Packages
Updated list of available packages in /data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_routing
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9/Packages
Updated list of available packages in /data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/openwrt_kiddin9
Downloading file:packages/Packages
Updated list of available packages in /data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../data/cache/c1/cache/mediatek/filogic/dl/imagebuilder
Collected errors:
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/luci-app-ddnsto_git-24.237.45156-121e6d1_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/ddnsto_3.0.4-29_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/luci-app-openclash_0.46.014-251_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/coreutils_9.3-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/coreutils-nohup_9.3-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/libcurl4_8.7.1-r1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/curl_8.7.1-r1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/libipset13_7.17-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/ipset_7.17-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/libbpf1_1.2.2-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-oxGqI6/ip-full_6.3.0-1_aarch64_cortex-a53.ipk': Operation not permitted. [... cut ...]
* opkg_conf_parse_file: Duplicate src declaration (openwrt_kiddin9 https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9). Skipping.
Pseudo file "dev" exists in source filesystem "/data/cache/c1/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/dev".
Ignoring, exclude it (-e/-ef) to override.
32486+1 records in
32487+0 records out
33266688 bytes (33 MB, 32 MiB) copied, 0.0779231 s, 427 MB/s
/data/cache/c1/cache/mediatek/filogic/staging_dir/host/bin/tar: Option --mtime: Treating date '@1719171470' as 2024-06-24 03:37:50

yixuan-ovo, 2024-06-24:
怎么刷进去的，用的opwrt官方的uboot吗？

beiyuqu, 2024-06-27:
同样的，360t7构建的固件也是没有无线