kiddin9/Kwrt #1781: 京东云亚瑟 AX1800 Pro 無法擴容

KardRi, 2024-06-21:
擴容分兩種，直接改分區別或改掛overlayfs到其他地方去。

```
umount /dev/mmcblk0p27
mkfs.ext4 -F /dev/mmcblk0p27
mkdir -p /mnt/mmcblk0p27
mount /dev/mmcblk0p27 /mnt/mmcblk0p27
cp -r /overlay/* /mnt/mmcblk0p27
ls /mnt/mmcblk0p27
block detect > /etc/config/fstab
sed -i s#/mnt/mmcblk0p27#/overlay# /etc/config/fstab
sed -i '12s/1/0/g' /etc/config/fstab
```

這個是可以成功把overlayfs掛到mmcblk0p27的，然而openwrtai載的固件沒辦法，不知道為什麼

比較istoreOs後，發現少裝了 blockd libmount1，但還是沒有解決

iStoreOS會去找
```
Sat Sep  2 01:10:43 2023 user.info kernel: [   11.910094] mount_root: loading kmods from internal overlay
Sat Sep  2 01:10:43 2023 user.info kernel: [   11.928200] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Sat Sep  2 01:10:43 2023 user.info kernel: [   11.929982] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Sat Sep  2 01:10:43 2023 user.info kernel: [   12.715671] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Sat Sep  2 01:10:43 2023 user.err kernel: [   12.715739] block: unable to load configuration (fstab: Entry not found)
Sat Sep  2 01:10:43 2023 user.info kernel: [   12.721451] block: attempting to load /tmp/overlay/etc/config/fstab
Sat Sep  2 01:10:43 2023 user.err kernel: [   12.728272] block: unable to load configuration (fstab: Entry not found)
Sat Sep  2 01:10:43 2023 user.info kernel: [   12.734210] block: attempting to load /etc/config/fstab
Sat Sep  2 01:10:43 2023 user.err kernel: [   12.742572] block: unable to load configuration (fstab: Entry not found)
Sat Sep  2 01:10:43 2023 user.err kernel: [   12.746117] block: no usable configuration
Sat Sep  2 01:10:43 2023 kern.info kernel: [   12.800198] EXT4-fs (loop0): recovery complete
Sat Sep  2 01:10:43 2023 kern.info kernel: [   12.800654] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
Sat Sep  2 01:10:43 2023 user.info kernel: [   12.803625] block: attempting to load /tmp/ext4_cfg/upper/etc/config/fstab
Sat Sep  2 01:10:43 2023 user.info kernel: [   12.812623] block: extroot: device not present, retrying in 5 seconds
Sat Sep  2 01:10:43 2023 user.notice kernel: [   18.516873] mount_root: switched to extroot
```

但openwrtai只會找
```
Sat Jun 22 02:46:40 2024 user.info kernel: [   14.456553] block: attempting to load /etc/config/fstab
Sat Jun 22 02:46:40 2024 user.err kernel: [   14.459131] block: unable to load configuration (fstab: Entry not found)
Sat Jun 22 02:46:40 2024 user.err kernel: [   14.460614] block: no usable configuration
Sat Jun 22 02:46:40 2024 kern.info kernel: [   14.468230] loop0: detected capacity change from 0 to 122880
Sat Jun 22 02:46:40 2024 kern.info kernel: [   14.535127] loop0: detected capacity change from 122880 to 8832
Sat Jun 22 02:46:40 2024 kern.info kernel: [   14.555858] EXT4-fs (loop0): recovery complete
Sat Jun 22 02:46:40 2024 kern.info kernel: [   14.556414] EXT4-fs (loop0): mounted filesystem with ordered data mode. Quota mode: disabled.
Sat Jun 22 02:46:40 2024 user.info kernel: [   14.561917] mount_root: loading kmods from internal overlay
Sat Jun 22 02:46:40 2024 user.info kernel: [   14.592924] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Sat Jun 22 02:46:40 2024 user.info kernel: [   14.593703] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Sat Jun 22 02:46:40 2024 user.info kernel: [   14.717259] block: attempting to load /etc/config/fstab
Sat Jun 22 02:46:40 2024 user.err kernel: [   14.717345] block: unable to load configuration (fstab: Entry not found)
Sat Jun 22 02:46:40 2024 user.err kernel: [   14.721317] block: no usable configuration
Sat Jun 22 02:46:40 2024 user.info kernel: [   14.728948] mount_root: switching to ext4 overlay
Sat Jun 22 02:46:40 2024 kern.warn kernel: [   14.735463] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.
```

這裡有提到null uuid，但我確認 /etc/config/fstab有把/overlay的entry對應的uuid改成了/dev/mmcblk0p27，但還是出現這個問題

KardRi, 2024-06-23:
終於搞定了，fstools要更新，但openwrt.ai會拉23.05的branch，再拉master的kernel，目前是6.1.94

拉完openwrt的branch後去改package/system/fstools/Makefile裡的PKG_MIRROR_HASH等以下三行

同時coolsnowwolf那邊的qca有對AX1800pro (改名叫re-ss-01?) 做最新支持，但我刷入download.openwrt.ai的re-ss-01會boot loop，只好把上個版本的patch拉回來，把diff修好(coolsnowwolf那邊有改內容，patch裡的diff要改一下)

```
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.100648] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Mon Jun 24 03:44:49 2024 user.err kernel: [   14.100760] block: unable to load configuration (fstab: Entry not found)
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.106593] block: attempting to load /tmp/overlay/etc/config/fstab
Mon Jun 24 03:44:49 2024 user.err kernel: [   14.113239] block: unable to load configuration (fstab: Entry not found)
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.119220] block: attempting to load /etc/config/fstab
Mon Jun 24 03:44:49 2024 user.err kernel: [   14.126253] block: unable to load configuration (fstab: Entry not found)
Mon Jun 24 03:44:49 2024 user.err kernel: [   14.131098] block: no usable configuration
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.138177] block: attempting to load /etc/config/fstab
Mon Jun 24 03:44:49 2024 user.err kernel: [   14.141958] block: unable to load configuration (fstab: Entry not found)
Mon Jun 24 03:44:49 2024 user.err kernel: [   14.147075] block: no usable configuration
Mon Jun 24 03:44:49 2024 kern.info kernel: [   14.154560] loop0: detected capacity change from 0 to 122880
Mon Jun 24 03:44:49 2024 kern.info kernel: [   14.214251] loop0: detected capacity change from 122880 to 94848
Mon Jun 24 03:44:49 2024 kern.info kernel: [   14.256599] EXT4-fs (loop0): recovery complete
Mon Jun 24 03:44:49 2024 kern.info kernel: [   14.257155] EXT4-fs (loop0): mounted filesystem with ordered data mode. Quota mode: disabled.
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.260307] mount_root: loading kmods from internal overlay
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.286819] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.287063] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.413687] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Mon Jun 24 03:44:49 2024 kern.info kernel: [   14.506900] EXT4-fs (mmcblk0p27): mounted filesystem with ordered data mode. Quota mode: disabled.
Mon Jun 24 03:44:49 2024 kern.warn kernel: [   14.511366] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.
Mon Jun 24 03:44:49 2024 kern.info kernel: [   14.517383] EXT4-fs (loop0): unmounting filesystem.
Mon Jun 24 03:44:49 2024 user.info kernel: [   14.527882] mount_root: switched to extroot
```

[image]

closed 2024-06-23 by KardRi

wgjwillis, 2024-07-06:
非常感谢你的修改和探索。希望大佬们能尽快同步代码！