kiddin9/Kwrt #1788: JDCloud AX1800 Pro固件与型号不对应，无法启动

master-yun-yun, 2024-07-01:
自动生成的固件与JDCloud AX1800 Pro不对应，自己建构也是对应生成的另外一个型号路由器的固件，导致无法使用无法自己建构，请修复

master-yun-yun, 2024-07-01:
选的ax1800 pro，
生成的是re-ss-01的固件，如图
[image]

master-yun-yun, 2024-07-03:
看图，最新版本依旧不对应

[image]

KardRi, 2024-07-03:
其實是同一個，但我有碰到bug，要拔wan才能正常的啟動，不然會沒反應(特徵是lan孔每大概3秒斷線一次，看起來是bootloop)

拔完wan後就能開了，再把wan插回去

master-yun-yun, 2024-07-04:
> 其實是同一個，但我有碰到bug，要拔wan才能正常的啟動，不然會沒反應(特徵是lan孔每大概3秒斷線一次，看起來是bootloop)
> 
> 拔完wan後就能開了，再把wan插回去
> 
> 至於為什麼要這樣取，coolsnowwolf那邊是這樣[取名](https://github.com/coolsnowwolf/lede/commit/a95c23c180dc48a7049bdf2a79b6bb5448b85034) 喔對了，從uboot刷，如果你分區是stock layout(沒特別刷過)要刷[recovery.bin](https://github.com/coolsnowwolf/lede/blob/7ecb40bb3a88b73ccb6314db126b1161661b632b/target/linux/qualcommax/image/ipq60xx.mk#L23)

我刷了无法启动。
意思是拔掉wan口线梦正常启动了？
灯一直红，也不闪。
我是wan口接主路由lan口。

master-yun-yun, 2024-07-04:
> 我是這麼發現的，從Uboot刷了recovery.bin後不插wan的情況下能正常地啟動
> 
> 至於wifi，如果是自己跑actions編譯，我有遇到編出來image裡的ipq-wifi不是對應的機種，可以看看log那邊有沒有正確的載入
> 
> LED燈就別理他了吧
> 
> 如果是預載了ipq-wifi-glinet_gl-ax1800，先把他移除再裝[ipq-wifi-jdcloud_ax1800pro](https://dl.openwrt.ai/releases/targets/qualcommax/ipq60xx/6.1.95/ipq-wifi-jdcloud_ax1800pro_2024-04-16-1d51799e-1_aarch64_cortex-a53.ipk)

刷了recovery.bin就能正常用了吗？不需要接着再刷别的？

master-yun-yun, 2024-07-07:
刷了recovery确实能用，但识别不了大分区，软件空间一点点儿，完全没法用。
放弃

KardRi, 2024-07-07:
> 刷了recovery确实能用，但识别不了大分区，软件空间一点点儿，完全没法用。 放弃

....，你去看我提的issue好了，你有改分區表嗎? 如果不是，就是把overlay搬家到後面那個大分區去。

由於kiddin9的是拉取23.05的原碼，僅從master branch替換kernel部分，而23.05的stable版的fstools有bug，搬overlay到後面那個p27區會無法正常跳過去啟動。我有提交新增一個patch丟在pull request，但作者不理我 lul

master-yun-yun, 2024-07-12:
在另外地方看见，也是拉的lede源码，要求先刷recovery，再sysupgrade。
难道这是lede源码的特殊要求？

KardRi, 2024-07-12:
> 在另外地方看见，也是拉的lede源码，要求先刷recovery，再sysupgrade。 难道这是lede源码的特殊要求？

不知道，純粹我是比對了 https://github.com/coolsnowwolf/lede/blob/7ecb40bb3a88b73ccb6314db126b1161661b632b/target/linux/qualcommax/image/ipq60xx.mk#L23

發現他的recovery.bin和別人的factory.bin都是pad到6144k，而他的factory.bin是pad到12288k

master-yun-yun, 2024-07-13:
> > 在另外地方看见，也是拉的lede源码，要求先刷recovery，再sysupgrade。 难道这是lede源码的特殊要求？
> 
> 不知道，純粹我是比對了 https://github.com/coolsnowwolf/lede/blob/7ecb40bb3a88b73ccb6314db126b1161661b632b/target/linux/qualcommax/image/ipq60xx.mk#L23
> 
> 發現他的recovery.bin和別人的factory.bin都是pad到6144k，而他的factory.bin是pad到12288k

 试了，先刷recovery，再刷sysupgrade是可以用的。
就是Lan口莫名其妙随机会变成100Mbps
再1个，无法开启ipv6
用istoreos的一键开启ipv6也没解决
[image]

master-yun-yun, 2024-07-14:
后来自己选插件建构了1个
再一键开启ipv6,重启,ipv6可以用了
随机变100mbps目前还没有重现,继续观察看看

master-yun-yun, 2024-07-19:
ipv6不稳定，有时可以，有时不行。
晚点儿试试7018版

master-yun-yun, 2024-07-20:
7018版无法DHCP
只能手动指定参数