kiddin9/Kwrt #1789: 似乎能获取到ipv6地址，但无法使用ipv6

zoope, 2024-07-03:
[image]
[image]

我用路由器直接拨号开启ipv6是能使用的，光猫应该没问题
没有什么头绪，该怎么排查？

KardRi, 2024-07-03:
你應該把wan_6擺到 wan的那個zone，把wan6那個刪除或先把他挪出wan的zone

sky540, 2024-07-04:
我最近我也发现这个问题

zoope, 2024-07-05:
> 你應該把wan_6(灰色的)擺到wan的那個zone(紅色的，與pppoe-wan同一個顏色)，把wan6(最下面那個)刪除或先把他挪出wan的zone，然後看你們是nat66還是有delegate可以用

不太清楚要怎么操作 我把wan6删了 并且在“设置向导”里把启用ipv6打开了 还需要别的设置吗

KardRi, 2024-07-05:
> > 你應該把wan_6(灰色的)擺到wan的那個zone(紅色的，與pppoe-wan同一個顏色)，把wan6(最下面那個)刪除或先把他挪出wan的zone，然後看你們是nat66還是有delegate可以用
> 
> 不太清楚要怎么操作 我把wan6删了 并且在“设置向导”里把启用ipv6打开了 还需要别的设置吗

選wan_6的編輯->防火牆設置->選紅色的wan

zoope, 2024-07-08:
> > > 你應該把wan_6(灰色的)擺到wan的那個zone(紅色的，與pppoe-wan同一個顏色)，把wan6(最下面那個)刪除或先把他挪出wan的zone，然後看你們是nat66還是有delegate可以用
> > 
> > 
> > 不太清楚要怎么操作 我把wan6删了 并且在“设置向导”里把启用ipv6打开了 还需要别的设置吗
> 
> 選wan_6的編輯->防火牆設置->選紅色的wan

我试过 但是这个wan_6无法编辑 只能去wan里面修改
[image]

zoope, 2024-07-08:
@KardRi 我搞定了 非常感谢
@sky540 你弄好了吗 我参照上面的思路 想办法把wan_6加到wan的防火墙zone里 
[image]
我是在高级设置里添加了`list network 'wan_6'`这一条 其他设置都不改，你也可以参考一下

closed 2024-07-08 by zoope