kiddin9/Kwrt #1793: 插件中心 tailscale 1.58.2-30 更新或者安装后不能使用

Raytheon-S, 2024-07-09:
插件中心的 tailscale 1.58.2-30 更新或者安装后不能使用，接口界面提示“缺失的设备”。

旧版本1.58.2-1可以正常使用，也可以tailscale update更新到官方同步1.68.2。

Powered by LuCI Master (git-25.158.56165-80b793c) / OpenWrt 23.05-SNAPSHOT (06.08.2024) / Lua

[image]

master-yun-yun, 2024-07-11:
反正主要是命令行，luci界面基本也干不了什么，不要也罢。
直接安装低版本，再tailscale update到官方最新版完事

Raytheon-S, 2024-07-12:
> 反正主要命令行，luci 界面基本也干什么，不要也罢。 安装低版本，再tailscale更新到最新版完事

谢谢，我通常用 opkg list-upgradable | cut -f 1 -d ' ' | xargs opkg upgrade 更新所有插件，所以会把tailscale更新到有问题的版本。找了具体的原因，这次插件列表的更新没有关联tailscaled这个依赖

sunnysu0608, 2024-07-13:
> 因

zerotier不香吗

master-yun-yun, 2024-07-14:
7月13的版本正常

cesaryuan, 2024-07-19:
确实存在问题，问题原因参考：#1801