kiddin9/Kwrt #1794: 请问 x86架构 如何调整根分区大小

CodeNewman, 2024-07-11:
如题，我使用istoreos可以实现上述，但是自定义x86镜像装好后，卡住了，fdisk显示磁盘剩余20G，但是df -h一直显示6G
```
Disk /dev/vda: 20 GiB, 21474836480 bytes, 41943040 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x3ecaff96

Device     Boot Start      End  Sectors Size Id Type
/dev/vda1  *     2048    34815    32768  16M 83 Linux
/dev/vda2       36864 41943039 41906176  20G 83 Linux
```
-
```
root@aliyun-online-003:~# df -h
Filesystem                Size      Used Available Use% Mounted on
tmpfs                   217.9M    136.0K    217.8M   0% /tmp
/dev/vda2                 6.0G     80.1M      5.9G   1% /
/dev/vda1                15.7M      7.2M      8.2M  47% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
```

master-yun-yun, 2024-07-11:
难道不是说20g是磁盘总空间？6g是一个分区的空间？

CodeNewman, 2024-07-22:
我后来想到解决方案了
再做一个分区，然后把openwrt的根分区复制到新分区中，
然后修改启动项目
然后从新的分区启动

zhangasd1, 2024-08-22:
> 我后来想到了解决方 再做一分区,当然,后分区的根区正在复兴, 当然后修目标 当然从新分区动画

您好，我在扩容overlay分区后重启又变成了默认的分区大小，请问有办法解决吗？感谢您的回复