kiddin9/Kwrt #1806: CMCC RAX3000M NAND 07-21 自定义构建固件无法启动，构建日志报错

laidu, 2024-07-21:
Building images for mediatek - CMCC RAX3000M NAND
Packages: iptables-nft luci-app-webadmin luci-ssl luci-theme-argon autocore base-files bash block-mount busybox ca-bundle coremark curl dnsmasq-full dropbear ds-lite e2fsprogs f2fsck firewall4 fitblk fstools htop kernel kmod-crypto-hw-safexcel kmod-gpio-button-hotplug kmod-leds-gpio kmod-lib-zstd kmod-mt7915e kmod-mt7981-firmware kmod-phy-aquantia kmod-tcp-bbr kmod-usb3 libc libgcc libustream-mbedtls logd luci-app-advancedplus luci-app-autoreboot luci-app-fan luci-app-fileassistant luci-app-firewall luci-app-opkg luci-app-upnp luci-app-wizard luci-base luci-compat luci-lib-fs luci-lib-ipkg mkf2fs mt7981-wo-firmware mtd nano netifd odhcp6c odhcpd-ipv6only openssh-sftp-server opkg ppp ppp-mod-pppoe procd procd-seccomp resolveip swconfig uboot-envtools uci uclient-fetch urandom-seed urngd wget-ssl wpad-basic-mbedtls zram-swap

Installing packages...
Installing libc (1.2.4-4) to root...
Installing libgcc1 (12.3.0-4) to root...
Downloading file:packages/libgcc1_12.3.0-4_aarch64_cortex-a53.ipk
Configuring libgcc1.
Configuring libc.
Installing kernel (6.6.40~1-r1) to root...
Configuring kernel.
Installing iptables-nft (1.8.8-2) to root...
Downloading file:packages/iptables-nft_1.8.8-2_aarch64_cortex-a53.ipk
Installing kmod-nf-reject (6.6.40-r1) to root...
Downloading file:packages/kmod-nf-reject_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nf-ipt (6.6.40-r1) to root...
Downloading file:packages/kmod-nf-ipt_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nf-log (6.6.40-r1) to root...
Downloading file:packages/kmod-nf-log_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-ipt-core (6.6.40-r1) to root...
Downloading file:packages/kmod-ipt-core_6.6.40-r1_aarch64_cortex-a53.ipk
Installing libmnl0 (1.0.5-1) to root...
Downloading file:packages/libmnl0_1.0.5-1_aarch64_cortex-a53.ipk
Installing libnftnl11 (1.2.6-12) to root...
Downloading file:packages/libnftnl11_1.2.6-12_aarch64_cortex-a53.ipk
Installing libxtables12 (1.8.8-2) to root...
Downloading file:packages/libxtables12_1.8.8-2_aarch64_cortex-a53.ipk
Installing libiptext0 (1.8.8-2) to root...
Downloading file:packages/libiptext0_1.8.8-2_aarch64_cortex-a53.ipk
Installing libiptext6-0 (1.8.8-2) to root...
Downloading file:packages/libiptext6-0_1.8.8-2_aarch64_cortex-a53.ipk
Installing libiptext-nft0 (1.8.8-2) to root...
Downloading file:packages/libiptext-nft0_1.8.8-2_aarch64_cortex-a53.ipk
Installing kmod-nfnetlink (6.6.40-r1) to root...
Downloading file:packages/kmod-nfnetlink_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nf-reject6 (6.6.40-r1) to root...
Downloading file:packages/kmod-nf-reject6_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nf-conntrack (6.6.40-r1) to root...
Downloading file:packages/kmod-nf-conntrack_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nf-conntrack6 (6.6.40-r1) to root...
Downloading file:packages/kmod-nf-conntrack6_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nf-nat (6.6.40-r1) to root...
Downloading file:packages/kmod-nf-nat_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nf-log6 (6.6.40-r1) to root...
Downloading file:packages/kmod-nf-log6_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-crypto-hash (6.6.40-r1) to root...
Downloading file:packages/kmod-crypto-hash_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-crypto-crc32c (6.6.40-r1) to root...
Downloading file:packages/kmod-crypto-crc32c_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-lib-crc32c (6.6.40-r1) to root...
Downloading file:packages/kmod-lib-crc32c_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nft-core (6.6.40-r1) to root...
Downloading file:packages/kmod-nft-core_6.6.40-r1_aarch64_cortex-a53.ipk
Installing kmod-nft-compat (6.6.40-r1) to root...
Downloading file:packages/kmod-nft-compat_6.6.40-r1_aarch64_cortex-a53.ipk
Installing xtables-nft (1.8.8-2) to root...
Downloading file:packages/xtables-nft_1.8.8-2_aarch64_cortex-a53.ipk
Installing luci-app-webadmin (git-24.217.56735-8015371) to root...
Downloading file://www/wwwroot/dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9/luci-app-webadmin_git-24.217.56735-8015371_all.ipk
Installing libubox20230523 (2023-05-23-75a3b870-1) to root...
Downloading file:packages/libubox20230523_2023-05-23-75a3b870-1_aarch64_cortex-a53.ipk
Installing libubus20230605 (2023-06-05-f787c97b-1) to root...
Downloading file:packages/libubus20230605_2023-06-05-f787c97b-1_aarch64_cortex-a53.ipk
Installing libuci20130104 (2023-08-10-5781664d-1) to root...
Downloading file:packages/libuci20130104_2023-08-10-5781664d-1_aarch64_cortex-a53.ipk
Installing libjson-c5 (0.16-3) to root...
Downloading file:packages/libjson-c5_0.16-3_aarch64_cortex-a53.ipk
Installing libblobmsg-json20230523 (2023-05-23-75a3b870-1) to root...
Downloading file:packages/libblobmsg-json20230523_2023-05-23-75a3b870-1_aarch64_cortex-a53.ipk
Installing rpcd (2023-07-01-c07ab2f9-1) to root...
[... 872 lines cut ...]
* copy_file: unable to preserve ownership of `/data/cache/c2/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-48O3ml/coreutils_9.3-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c2/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-48O3ml/coreutils-stat_9.3-1_aarch64_cortex-a53.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c2/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-48O3ml/luci-i18n-base-zh-cn_git-24.180.72161-85dec60_all.ipk': Operation not permitted.
* opkg_install_cmd: Cannot install package luci-i18n-base-en.
Collected errors:
* opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/base). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/packages). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/luci). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/routing). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_kiddin9 https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9). Skipping.
* copy_file: unable to preserve ownership of `/data/cache/c2/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-kvVwzs/luci-i18n-firewall-zh-cn_git-24.180.72161-85dec60_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c2/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-kvVwzs/luci-i18n-opkg-zh-cn_git-24.180.72161-85dec60_all.ipk': Operation not permitted.
* copy_file: unable to preserve ownership of `/data/cache/c2/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/tmp/opkg-kvVwzs/luci-i18n-upnp-zh-cn_git-24.180.72161-85dec60_all.ipk': Operation not permitted.
Collected errors:
* opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/base). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/packages). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/luci). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/routing). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_kiddin9 https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9). Skipping.
Pseudo file "dev" exists in source filesystem "/data/cache/c2/cache/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/dev".
Ignoring, exclude it (-e/-ef) to override.
11015+1 records in
11016+0 records out
11280384 bytes (11 MB, 11 MiB) copied, 0.0259656 s, 434 MB/s
/data/cache/c2/cache/mediatek/filogic/staging_dir/host/bin/tar: Option --mtime: Treating date '@1721330598' as 2024-07-19 03:23:18

xibeilang0616, 2024-07-21:
同一个型号，本人的完美。估计你是刷的某论坛大神提供的uboot，这些野路子uboot，七改八改，兼容性容易出大问题，变砖的可能性极大。
最好重刷openwrt官网的uboot。
https://firmware-selector.openwrt.org/

laidu, 2024-07-21:
> 同一个型号，本人的完美。估计你是刷的某论坛大神提供的uboot，这些野路子uboot，七改八改，兼容性容易出大问题，变砖的可能性极大。 最好重刷openwrt官网的uboot。 https://firmware-selector.openwrt.org/

uboot使用的是 https://github.com/hanwckf/bl-mt798x
同uboot下，07.05定制固件启动正常

xibeilang0616, 2024-07-21:
> > 同一个型号，本人的完美。估计你是刷的某论坛大神提供的uboot，这些野路子uboot，七改八改，兼容性容易出大问题，变砖的可能性极大。 最好重刷openwrt官网的uboot。 https://firmware-selector.openwrt.org/
> 
> uboot使用的是 https://github.com/hanwckf/bl-mt798x 同uboot下，07.05定制固件启动正常

还有一个提醒，编译时不要太贪功能多，对于RAX3000M最好控制在100兆以内，50兆左右最佳，其他复杂的功能该机型也干不了，联发科芯片擅长网络，不擅长安装什么NAS功能

KardRi, 2024-07-21:
不知道這位的版本有沒有支持nmbm

laidu, 2024-07-21:
> > > 同一个型号，本人的完美。估计你是刷的某论坛大神提供的uboot，这些野路子uboot，七改八改，兼容性容易出大问题，变砖的可能性极大。 最好重刷openwrt官网的uboot。 https://firmware-selector.openwrt.org/
> > 
> > 
> > uboot使用的是 https://github.com/hanwckf/bl-mt798x 同uboot下，07.05定制固件启动正常
> 
> 还有一个提醒，编译时不要太贪功能多，编译时添加自己熟悉的插件，精简掉自己不熟悉，不知道干什么玩意的插件，对于RAX3000M最好控制在100兆以内，本身存储才128M，刷100兆的固件可能瞬间把128M存储塞满而且有可能不够出无法启动系统的故障。 50兆上下最佳，其他复杂的功能该机型也干不了，联发科芯片擅长网络路由专业，不擅长安装什么NAS、下载等功能，我测试过BT下载功能，芯片被打爆100％卡机。

默认配置、无额外增加软件，定制编译依然报错，固件无法启动

KardRi, 2024-07-21:
> 默认配置、无额外增加软件，定制编译依然报错，固件无法启动

他最新的patch有改了filogic/base-files/etc/board.d/02_network，你再試試看吧