kiddin9/Kwrt #1809: 玩客云luci-app-daed无法启动，内核选项不全，请求更新

FreeJ233, 2024-07-23:
现有内核选项
CONFIG_BPF=y
CONFIG_BPF_SYSCALL=y
CONFIG_BPF_JIT=y
CONFIG_CGROUPS=y
CONFIG_NET_INGRESS=y
CONFIG_NET_EGRESS=y
CONFIG_NET_SCH_INGRESS=m
CONFIG_NET_CLS_BPF=m
CONFIG_NET_CLS_ACT=y
CONFIG_DEBUG_INFO=y

大鹅dae的内核要求选项
CONFIG_BPF=y
CONFIG_BPF_SYSCALL=y
CONFIG_BPF_JIT=y
CONFIG_CGROUPS=y
CONFIG_KPROBES=y
CONFIG_NET_INGRESS=y
CONFIG_NET_EGRESS=y
CONFIG_NET_SCH_INGRESS=m
CONFIG_NET_CLS_BPF=m
CONFIG_NET_CLS_ACT=y
CONFIG_BPF_STREAM_PARSER=y
CONFIG_DEBUG_INFO=y
# CONFIG_DEBUG_INFO_REDUCED is not set
CONFIG_DEBUG_INFO_BTF=y
CONFIG_KPROBE_EVENTS=y
CONFIG_BPF_EVENTS=y
CONFIG_BPF_JIT_DEFAULT_ON=y

缺少
CONFIG_KPROBES=y
CONFIG_BPF_STREAM_PARSER=y
# CONFIG_DEBUG_INFO_REDUCED is not set
CONFIG_DEBUG_INFO_BTF=y
CONFIG_KPROBE_EVENTS=y
CONFIG_BPF_EVENTS=y
CONFIG_BPF_JIT_DEFAULT_ON=y

请求更新

isvicy, 2024-10-15:
终身会员请求更新 +1

FreeJ233, 2024-11-01:
> 终身会员请求更新 +1

已放弃，哥们别想了，我查了很多资料，也自编译了很多版本，发现是底层arm32位linux内核的问题，这个问题之前arm64位也有，不过现在修复了，看看有没有大佬可以修复32位的内核吧

anyUesr, 2024-12-16:
终身会员请求更新 +1