kiddin9/Kwrt #1810: CMCC RAX3000M NAND 22日编译的固件无法刷机

xibeilang0616, 2024-07-23:
**###** 今天闲来无事，下载最新的固件刷机升级，结果刷进去无法重启，平时自动重启时红灯闪烁，稍后绿灯表示重启成功。今天不行，蓝灯闪两下后，一直时蓝灯，无法进入系统。
没办法找了一个老版ImmortalWrt刷机，刷机成功，凑活用。
怀疑是uboot没升级？到openwrt官方找23.05.4 的uboot刷好，再刷贵站的openwrt，还是刷进去不重启，进不来系统。最后一次退回刷ImmortalWrt
出了这么大的问题？

wujun4code, 2024-07-31:
@xibeilang0616 https://firmware-selector.immortalwrt.org/?version=23.05.3&target=mediatek%2Ffilogic&id=cmcc_rax3000m 这个可以刷进去，我试过了，这是 23.05 最新版了。

wujun4code, 2024-07-31:
@xibeilang0616 但是还是 openwrt.ai 生成的固件最好用，实在没时间细研究为啥刷不了 openwrt.ai 生成的 rax3000m 的固件，我用 nand 和 emmc 版本都试过了，就是无法启动。

xibeilang0616, 2024-07-31:
> @xibeilang0616 https://firmware-selector.immortalwrt.org/?version=23.05.3&target=mediatek%2Ffilogic&id=cmcc_rax3000m 这个可以刷进去，我试过了，这是 23.05 最新版了。

### **谢谢，真个版本是能刷，不过这个版本维护太落后，好像倒闭的样子，最重要固件软件库里软件太少，大部分搜不出来**

wujun4code, 2024-07-31:
@xibeilang0616 因为它是 immortalwrt 最最最最精简的版本，并且内核还不能指定，内核是奇怪的 5.15.162，这个版本那对应的 kmod 内核库一大堆组件在 几乎所有的软件源上无法下载对应版本的组件，比如就装不上 luci-app-quickstart。。。