kiddin9/Kwrt #1818: TP-Link TL-XDR6088用debug固件刷机6.6版本内核不成功

AnChiehChen, 2024-07-27:
TP-Link TL-XDR6088用debug固件刷6.6内核版本不成功，试了几个版本一个也刷不上去， 谷歌浏览器清理缓存。显示固件上传成功后，自动获取的ip是169开头的，请问怎么回事呢？能处理一下吗？

xibeilang0616, 2024-07-27:
大面积的故障，不是你一个人，你刷7月13日历史版本，应该能刷进去

AnChiehChen, 2024-07-27:
> 大面积的故障，不是你一个人，你刷7月13日历史版本，应该能刷进去

7月13日历史版本，你用debug固件刷成功了吗？

xibeilang0616, 2024-07-27:
> > 大面积的故障，不是你一个人，你刷7月13日历史版本，应该能刷进去
> 
> 7月13日历史版本，你用debug固件刷成功了吗？

现在用的就是这个，你试一试

AnChiehChen, 2024-07-27:
> > > 大面积的故障，不是你一个人，你刷7月13日历史版本，应该能刷进去
> > 
> > 
> > 7月13日历史版本，你用debug固件刷成功了？
> 
> 现在用的就是这个，你试一试

你是debug固件下刷成功的？还是debug固件下先刷过度固件，然后在过度固件的基础上再刷7.13号的版本才成功的？

xibeilang0616, 2024-07-27:
> > > > 大面积的故障，不是你一个人，你刷7月13日历史版本，应该能刷进去
> > > 
> > > 
> > > 7月13日历史版本，你用debug固件刷成功了？
> > 
> > 
> > 现在用的就是这个，你试一试
> 
> 你是debug固件下刷成功的？还是debug固件下先刷过度固件，然后在过度固件的基础上再刷7.13号的版本才成功的？

直接uboot刷的7.13

AnChiehChen, 2024-07-27:
> 直接uboot刷的7.13

用uboot刷的和debug刷的，文件格式不一样的，我一会试试吧；

he110777, 2024-08-01:
6086的也是一样，启动不了，怀疑是网卡驱动没有

he110777, 2024-08-01:
> 大面积的故障，不是你一个人，你刷7月13日历史版本，应该能刷进去

历史版本怎么只有200.7 KiB那个文件

AnChiehChen, 2024-08-13:
> 6086的也是一样，启动不了，怀疑是网卡驱动没有

没有网卡驱动那更新个什么，更新的又刷不上去，无语的很；

AnChiehChen, 2024-08-13:
> > 大面积的故障，不是你一个人，你刷7月13日历史版本，应该能刷进去
> 
> 历史版本怎么只有200.7 KiB那个文件

没有那么小的文件吧

li1553770945, 2024-09-26:
遇到了同样的问题，刷进去之后没有无线，也没有DHCP服务器

AnChiehChen, 2024-09-26:
刷不上去的 我也试过

AnChiehChen, 2024-11-25:
> 遇到了同样的问题，刷进去后没有无线，也没有DHCP服务器

你在debua固件下刷的吧？