kiddin9/Kwrt #1827: n1刷进去之后，查看接口wifi等页面反应特别慢

LoadingZhang, 2024-08-05:


Tim-tse, 2024-08-05:
最近这版有问题 约15秒才能正常加载完成，等官方修复吧。我刷F大的90+o 90+也有这个问题，F大5月份那个包89+o 89+、从加载到显示约3秒左右。

LoadingZhang, 2024-08-05:
https://github.com/openwrt/openwrt/issues/14013
我看到这个issue里提到和wifi有关，在其他设备上已经修复了。
我把wifi关了问题就没了。
cc @Tim-tse

Tim-tse, 2024-08-06:
> [openwrt/openwrt#14013](https://github.com/openwrt/openwrt/issues/14013) brcmfmac: brcmf_dump_obss: dump_obss error (-52) 我看到这个问题里提到和wifi有关，在其他设备上已经修复了。 我把wifi关了问题也没有了 。cc@Tim-tse

还真是，禁用后正常了。我一开始以为是iStoreOS问题，后来发现不是他的原因。

ooyyee, 2024-08-06:
+1,继续使用老版本，目前只能生成6.6版本的了。

YEUNGWAIPANG, 2024-08-19:
> > [openwrt/openwrt#14013](https://github.com/openwrt/openwrt/issues/14013) brcmfmac: brcmf_dump_obss: dump_obss error (-52) 我看到这个问题里提到和wifi有关，在其他设备上已经修复了。我把wifi关了问题也没有了。cc@Tim-tse
> 
> 果然，禁用后正常了。我一开始以为是iStoreOS的问题，后来发现不是他的原因。

请问一下大佬，把wifi禁用是直接在无线中禁用wifi吗？我这样子做还是不行。请问能提供一下操作截图吗？

Tim-tse, 2024-08-21:
[image]
目前继续使用老版本,或这样关闭AP。