kiddin9/Kwrt #1836: openwrt 更新后 Bypass出面一些小BUG

NianCai, 2024-08-17:
将openwrt更新到 OpenWrt 07.26.2024 by Kiddin' / LuCI Master git-25.206.62814-57777b9 后 翻墙插件bypass出现无法获取地址及旁边图标不能点亮的问题，bypass组件Xray版本为1.8.23

gaifeng8864, 2024-08-22:
> 将openwrt更新到 OpenWrt 07.26.2024 by Kiddin' / LuCI Master git-25.206.62814-57777b9 后 翻墙插件bypass出现无法获取地址及旁边图标不能点亮的问题，bypass组件Xray版本为1.8.23

我也注意到这个问题了，在最新的20240822发布的版本中依旧存在这个问题。我的设备：
型号:	Redmi AX6
平台:	ipq807x/generic
版本:	分支: 23.05-SNAPSHOT
内核: 5.15.164
生成日期:	2024-08-22 12:59:55

老兄，你的这个问题解决了吗？

NianCai, 2024-08-24:
> > 将openwrt更新到 OpenWrt 07.26.2024 by Kiddin' / LuCI Master git-25.206.62814-57777b9 后 翻墙插件bypass出现无法获取地址及旁边图标不能点亮的问题，bypass组件Xray版本为1.8.23
> 
> 我也注意到这个问题了，就是Bypass主界面最下方的地址获取栏总是地球在旋转，一直显示“获取中。。。”但是其它功能都正常，对吧？在最新的20240822发布的版本中依旧存在这个问题。我的设备： 型号： Redmi AX6 平台： ipq807x/generic 版本： 分支： 23.05-SNAPSHOT 内核： 5.15.164 生成日期： 2024-08-22 12：59：55
> 
> 老兄，你的这个问题解决了吗？

没有解决，不管了其他能正常使用

gaifeng8864, 2024-08-25:
> 没有解决，不管了其他能正常使用

感谢回复。不过，经过我长时间多次更换版本测试，目前最好用的还是 20240705发布的内核为6.6.35的版本。这个版本固件把无线驱动换成20230331的那版后速度非常快。最新版的固件经过测试除了bypass不能显示地址外，网络速度总是差点意思。

NianCai, 2024-08-27:
> 感谢回复。不过，经过我长时间多次更换版本测试，目前最好用的还是 20240705发布的内核为6.6.35的版本。这个版本固件把无线驱动换成20230331的那版后速度非常快。最新版的固件经过测试除了bypass不能显示地址外，网络速度总是差点意思。

是吗，我有时间试试

NianCai, 2024-09-03:
> 感谢回复。不过，经过我长时间多次更换版本测试，目前最好用的还是 20240705发布的内核为6.6.35的版本。这个版本固件把无线驱动换成20230331的那版后速度非常快。最新版的固件经过测试除了bypass不能显示地址外，网络速度总是差点意思。

[image]
我将Openwrt更新到最新版本 然后bypass也更新到最新的版本后 就解决了  我看了一下xray组件还是更新了 应该是这个更新解决了这个BUG了

gaifeng8864, 2024-09-03:
> > > > > 将openwrt更新到 OpenWrt 07.26.2024 by Kiddin' / LuCI Master git-25.206.62814-57777b9 后 翻墙插件bypass出现无法获取地址及旁边图标不能点亮的问题，bypass组件Xray版本为1.8.23
> > > > 
> > > > 
> > > > 我也注意到这个问题了，就是Bypass主界面最下方的地址获取栏总是地球在旋转，一直显示“获取中。。。”但是其它功能都正常，对吧？在最新的20240822发布的版本中依旧存在这个问题。我的设备： 型号： Redmi AX6 平台： ipq807x/generic 版本： 分支： 23.05-SNAPSHOT 内核： 5.15.164 生成日期： 2024-08-22 12：59：55
> > > > 老兄，你的这个问题解决了吗？
> > > 
> > > 
> > > 没有解决，不管了其他能正常使用
> > 
> > 
> > 感谢回复。不过，经过我长时间多次更换版本测试，目前最好用的还是 20240705发布的内核为6.6.35的版本。这个版本固件把无线驱动换成20230331的那版后速度非常快。最新版的固件经过测试除了bypass不能显示地址外，网络速度总是差点意思。
> 
> ![2024-09-03](https://private-user-images.githubusercontent.com/154070798/363853898-dbc1133d-cc22-445c-9d02-16bf085e97d1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNDY5MTMsIm5iZiI6MTcyNTM0NjYxMywicGF0aCI6Ii8xNTQwNzA3OTgvMzYzODUzODk4LWRiYzExMzNkLWNjMjItNDQ1Yy05ZDAyLTE2YmYwODVlOTdkMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QwNjU2NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZWFmYzc5NzQ4MjNhNjkyMmExMTA4MWRkYWE0NGIxNDlhNTljODY0Y2Y3ZTBiNjkwM2IwNjM4YWQwOWE1YWFiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1KMbfrvn6AbF5kbw3CJprp3tU-KtxjIqibib8Uf-JJw) 我将Openwrt更新到最新版本 然后bypass也更新到最新的版本后 就解决了 我看了一下xray组件还是更新了 应该是这个更新解决了这个BUG了

我也更新了测试了，和你一样也解决了这个bug，应该这个版本修复了。感谢~