kiddin9/Kwrt #1841: 7.26以后的新版本快捷访问不可用，希望可以重新添加这个功能

kingfingerpro, 2024-08-22:
检查脚本之后发现是新版本修改了 `/etc/init.d/dnsmasq` ，删除了一段代码导致的
```bash
	network_get_ipaddr lanaddr "lan"
	if [ -n $lanaddr ]; then
 	lanaddr="$(echo "$lanaddr" | sed -e "s/\/24//")"
	echo "
	$lanaddr op" >/tmp/hosts/shortcuts
	if [ -f /etc/nginx/conf.d/shortcuts.conf ]; then
	if [[ "$lanaddr" != "`grep "set \$ip" /etc/nginx/conf.d/shortcuts.conf | head -n1 | cut -f 3 -d " " | sed -e 's/;//'`" ]]; then
		sed -i "/set \$ip/c\set \$ip $lanaddr;" /etc/nginx/conf.d/shortcuts.conf
		/etc/init.d/nginx reload
	fi
	cat /etc/nginx/conf.d/shortcuts.conf | tr "\n" "\r" |grep -oE 'server_name [a-z.-]*;[^;]*set \$ip [0-9.]*;' | sed -e 's/server_name \(.*\);.*location.*set $ip \(.*\);/\2 \1/' >>/tmp/hosts/shortcuts
	cat /etc/nginx/conf.d/shortcuts.conf | tr "\n" "\r" |grep -oE 'server_name [a-z.-]*;[^;]*return 301 \$scheme://op' | sed -e 's/server_name \(.*\?\);.*/\1/' | sed -e "s/^/$lanaddr /">>/tmp/hosts/shortcuts
	fi
	fi
```
[image]
希望大佬能把这个功能加回来

ni2hao4, 2025-02-04:
对，我也发现了github上提到的快捷访问在这个目录地址下，/etc/nginx/conf.d/shortcuts.conf  但实际上找不到，虽然可以到系统设置向导里找到，但是每次保存以后，wan口会重连，导致短暂断网和ddns也要等待刷新，大佬看看能不能给加一个独立的入口设置呢？

谢谢🙏

kingfingerpro, 2025-02-04:
> 对，我也发现了github上提到的快捷访问在这个目录地址下，/etc/nginx/conf.d/shortcuts.conf 但实际上找不到，虽然可以到系统设置向导里找到，但是每次保存以后，wan口会重连，导致短暂断网和ddns也要等待刷新，大佬看看能不能给加一个独立的入口设置呢？
> 
> 谢谢🙏

我现在的做法是每次更新完之后手动在`/etc/init.d/dnsmasq`的老位置重新插入被删除的脚本，你可以用下面的，我改了下正则让脚本可以匹配包含数字的server_name（比如g006.le）

```bash

	network_get_ipaddr lanaddr "lan"
	if [ -n $lanaddr ]; then
 	lanaddr="$(echo "$lanaddr" | sed -e "s/\/24//")"
	echo "
	$lanaddr op" >/tmp/hosts/shortcuts
	if [ -f /etc/nginx/conf.d/shortcuts.conf ]; then
	if [[ "$lanaddr" != "`grep "set \$ip" /etc/nginx/conf.d/shortcuts.conf | head -n1 | cut -f 3 -d " " | sed -e 's/;//'`" ]]; then
		sed -i "/set \$ip/c\set \$ip $lanaddr;" /etc/nginx/conf.d/shortcuts.conf
		/etc/init.d/nginx reload
	fi
	cat /etc/nginx/conf.d/shortcuts.conf | tr "\n" "\r" |grep -oE 'server_name [a-zA-Z0-9.-]*;[^;]*set \$ip [0-9.]*;' | sed -e 's/server_name \(.*\);.*location.*set $ip \(.*\);/\2 \1/' >>/tmp/hosts/shortcuts
	cat /etc/nginx/conf.d/shortcuts.conf | tr "\n" "\r" |grep -oE 'server_name [a-zA-Z0-9.-]*;[^;]*return 301 \$scheme://op' | sed -e 's/server_name \(.*\?\);.*/\1/' | sed -e "s/^/$lanaddr /">>/tmp/hosts/shortcuts
	fi
	fi

```