kiddin9/Kwrt #1845: HelloWorld本来很好用,不知道为什么现在全部替换成现在这些了,很难用

joesdu, 2024-08-24:
HelloWorld本来很好用,不知道为什么现在全部替换成现在这些了,很难用,经常出现无法分流,或者代理无效.有些特别的卡很慢.

Doctorlee0827, 2024-08-30:
防火墙换成firewall4了，有些软件还没完全适配，用passwall即可

joesdu, 2024-08-30:
> 防火墙换成firewall4了，有些软件还没完全适配，用passwall即可

passwall我这边根本起不来，没有任何效果，酸酸乳+ 分流效果很有问题，经常打开国内网站走了代理，要么打不开，要么特别卡，还有bing经常无法访问，目前只能用了clash才能勉强用的样子，不过clash老是出新闻，有点不太放心😂

Doctorlee0827, 2024-08-30:
> > 防火墙换成firewall4了，有些软件还没完全适配，用passwall即可
> 
> passwall我这边根本起不来，没有任何效果，酸酸乳+ 分流效果很有问题，经常打开国内网站走了代理，要么打不开，要么特别卡，还有bing经常无法访问，目前只能用了clash才能勉强用的样子，不过clash老是出新闻，有点不太放心😂

重装试试，升级问题很多，重新安装就好了

joesdu, 2024-09-02:
> > > 防火墙换成firewall4了，有些软件还没完全适配，用passwall即可
> > 
> > 
> > passwall我这边根本起不来，没有任何效果，酸酸乳+ 分流效果很有问题，经常打开国内网站走了代理，要么打不开，要么特别卡，还有bing经常无法访问，目前只能用了clash才能勉强用的样子，不过clash老是出新闻，有点不太放心😂
> 
> 重装试试，升级问题很多，重新安装就好了

试过好几次了,还是不行.不过OpenClash的资源消耗太多了10% 往上的CPU占用.

closed 2024-09-02 by joesdu