kiddin9/Kwrt #1856: nonopi的这个升级bug这么多版本了一直没有修复

nonozone, 2024-09-02:
[image]
nanopi升级的时候，总是提示找不到镜像文件，实际上是系统的这个更新文件的文件名不对，后缀应该改成img.gz

Duneon, 2024-10-04:
可以看一下你在线升级配置的服务器地址么？我的完全无法升级，点了没反应

nonozone, 2024-10-09:
> 可以看一下你在线升级配置的服务器地址么？我的完全无法升级，点了没反应

`src/gz kwrt_core https://dl.openwrt.ai/23.05/targets/rockchip/armv8/6.6.47
src/gz kwrt_base https://dl.openwrt.ai/23.05/packages/aarch64_generic/base
src/gz kwrt_packages https://dl.openwrt.ai/23.05/packages/aarch64_generic/packages
src/gz kwrt_luci https://dl.openwrt.ai/23.05/packages/aarch64_generic/luci
src/gz kwrt_routing https://dl.openwrt.ai/23.05/packages/aarch64_generic/routing
src/gz kwrt_kiddin9 https://dl.openwrt.ai/23.05/packages/aarch64_generic/kiddin9
`