kiddin9/Kwrt #1861: R4S从去年好像8月份以后就不能用

chenyajunai, 2024-09-13:
R4S从去年好像8月份以后就不能用，下载默认的不是定制的也启动不起来，SYS灯不亮

ckyd198613, 2024-09-25:
同问，r4s不更新了吗？

Duneon, 2024-10-01:
R4S可以用啊，我的正常，只是在线升级已经很久不管用了，发邮件反馈也没回音，估计放弃维护了吧？

chenyajunai, 2024-10-02:
> R4S可以用啊，我的正常，只要在线升级已经不管用了，发邮件反馈也没回音，估计够维持了吧？

我的用不了，目前只能用之前保留的23年7月份的一个固件

Duneon, 2024-10-05:
> > R4S可以用啊，我的正常，只要在线升级已经不管用了，发邮件反馈也没回音，估计够维持了吧？
> 
> 我的用不了，目前只能用之前保留的23年7月份的一个固件

你是用什么制作的系统盘，rufus么？我昨天折腾了一下，刷了好几次的istoreos和这个（固件用的最新的），也出现过一次灯不亮，我换了张TF卡，就好了，但是刷进去系统有些不正常，后面我又换balenaEtcher制作系统盘，目前一切正常（除了文件管理无法上传文件，提示上传失败，请稍后再试，这个系统自带的文件管理太拉了，以前还出现过页面不显示，我现在用易有云代替）。

chenyajunai, 2024-10-05:
我一直用balenaEtcher刷入，因为之前出现过sys灯不亮，系统进不去也用rufus试过依旧不行，内存卡也换过还是不行，最近好长时间都没有定制了，还是刷回到了之前保留的23年7月底的版本了

jacobleft, 2024-12-18:
已经解决了，可以先从老版本刷进去，然后再升级到最新版。直接把最新版固件烧进TF卡无法进系统，这个很奇怪。

_Originally posted by @DeadpoolRain in https://github.com/istoreos/istoreos/issues/1106#issuecomment-1808220877_
            
这个方法实测可行，我用的分别是07.06.2024和12.04.2024非定制版；

[image]

zhizhizz, 2025-01-05:
> ```
>           已经解决了，可以先从老版本刷进去，然后再升级到最新版。直接把最新版固件烧进TF卡无法进系统，这个很奇怪。
> ```
> 
> _Originally posted by @DeadpoolRain in [istoreos/istoreos#1106 (comment)](https://github.com/istoreos/istoreos/issues/1106#issuecomment-1808220877)_
> 
> 这个方法实测可行，我用的分别是07.06.2024和12.04.2024非定制版；
> 
> ![image](https://private-user-images.githubusercontent.com/88961633/396810423-f7f02563-c196-462d-8e88-18bf01c18fb9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNTg1NzUsIm5iZiI6MTczNjA1ODI3NSwicGF0aCI6Ii84ODk2MTYzMy8zOTY4MTA0MjMtZjdmMDI1NjMtYzE5Ni00NjJkLThlODgtMThiZjAxYzE4ZmI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDA2MjQzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5ZGExMDA4OGM0MTIyZThkMzhiZjc5NzFiNTA3ZTAyN2Y2MWNlNDYzMGZmM2E4MzdjMDBjMThhNmVmOWQ5OTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.FRXTfcajmNlvLd9f5Te2EDowa4ITumeULDzykYCsOyg)

这个确实测试有效