kiddin9/Kwrt #1875: 9月26日的固件编译，会导致无法进入路由器管理页面

CharlieVSAngel, 2024-09-29:
在https://openwrt.ai/上编译的Generic x86/64和FriendlyARM NanoPi R5S的固件，无法登录http://192.168.1.1/页面。发现是系统没有uhttpd插件，解决方法是：
通过SSH连接后台，
1.opkg update;
2.opkg install uhttpd;
3.运行 /etc/init.d/uhttpd start 或者 /etc/init.d/uhttpd restart。

就可以解决！

希望大佬下次固件更新时候解决。

klpk, 2024-10-01:
我还是没没能成功打开
Installing uhttpd (2023-06-25-34a8a74d-2) to root...
Downloading https://dl.openwrt.ai/23.05/packages/aarch64_generic/base/uhttpd_2023-06-25-34a8a74d-2_aarch64_generic.ipk
Configuring uhttpd.
4+0 records in
4+0 records out

CharlieVSAngel, 2024-10-02:
10月2日的固件编译依然无法进入路由器管理页面

Albert9364, 2024-10-03:
NanoPi R2S也是这样，用着用着就崩了，路由器页面进不去，网页解析不了，只能断电重启

laiyoi, 2024-10-05:
一样，但是ping都ping不通

AbsurdYu, 2024-10-07:
9.26 的 R2S 固件，配置了 adguardhome 和 pass wall 后，passwall 日志中显示防火墙重启，大概率卡死，下游的 AP 硬路由红灯连接不上，使用网线直连也分配不到正确的 ip，然后就会重启死循环，重刷后尝试不配置 adg，只使用 passwall，追加规则列表后保存应用，进入防火墙重启阶段也有概率陷入死循环导致无法使用，怀疑是新版防火墙问题，没精力去验证😂

CharlieVSAngel, 2024-10-17:
> 9.26 的 R2S 固件，配置了 adguardhome 和 pass wall 后，passwall 日志中显示防火墙重启，大概率卡死，下游的 AP 硬路由红灯连接不上，使用网线直连也分配不到正确的 ip，然后就会重启死循环，重刷后尝试不配置 adg，只使用 passwall，追加规则列表后保存应用，进入防火墙重启阶段也有概率陷入死循环导致无法使用，怀疑是新版防火墙问题，没精力去验证😂

最近一直在测试，感觉9月份后的固件都可能会因为防火墙导致无法连接管理页面