kiddin9/Kwrt #1878: 无法在线更新

Ikeltis, 2024-10-03:
[image]
qing'jiao'xia
请教下这是为什么，直接复制这个链接打开也是404，[https://dl.openwrt.ai/firmware/x86_64/](url)这个链接可以打开，点进去也能下载。

kukudemajia, 2024-10-29:
文件下载失败或超时：https://dl.openwrt.ai/firmware/x86_64/09.26.2024-openwrt-x86-64-generic-squashfs-combined.img.gz
同样的问题

hisune, 2024-12-01:
大佬们有解决方案吗？我也有这个问题，已经好久没法更新了

kukudemajia, 2024-12-21:
> 大佬们有解决方案吗？我也有这个问题，已经好久没法更新了

直接修改/usr/lib/lua/luci/model/cbi/gpsysupgrade/sysupgrade.lua

fivedang, 2025-03-27:
> > 大佬们有解决方案吗？我也有这个问题，已经好久没法更新了
> 
> 直接修改/usr/lib/lua/luci/model/cbi/gpsysupgrade/sysupgrade.lua

我来补充一下这个回复。
出错原因：固件的文件命名规则改了，固件中的openwrt被改为kwrt。（注：经过查询https://dl.openwrt.ai/firmware/x86_64/目录可获得最新的固件的名称）
解决方案：修改/usr/lib/lua/luci/model/cbi/gpsysupgrade/sysupgrade.lua这个文件，将设置中文件名中的openwrt改为kwrt。

改完之后直接再运行在线更新就行。