kiddin9/Kwrt #1909: 自定义固件防火墙界面无法加载

laiyoi, 2024-10-26:
[image]

kittyscience, 2024-10-27:
哟西

ITCJ, 2024-10-27:
遇到了同样的问题

lvxj11, 2024-10-27:
同样问题

laiyoi, 2024-11-02:
问题找到了，浏览器调试对`form.RichListValue`的显示为undefined
[image]
同时在kwrt-packages/commit/9ab308中发现此部分被修改
[image]

laiyoi, 2024-11-02:
猜测更新luci-app-firewall可解决

laiyoi, 2024-11-02:
已使用最新的luci-app-firewall，仍无效
[image]

laiyoi, 2024-11-02:
在更新最新luci-app-firewall并重启之后js代码仍为下图
[image]
而`/www/luci-static/resources/view/firewall`中已改变
[image]
有什么更新的方法吗？

laiyoi, 2024-11-02:
新发现，bootstrap主题可解决
[image]
但是切回argon主题仍不行

laiyoi, 2024-11-02:
在bootstrap主题下zones.js的内容也更新为正确版本
[image]

laiyoi, 2024-11-02:
神奇，CTRL+shift+r解决

laiyoi, 2024-11-02:
推测：需更新或删除软路由内部类似缓存的文件