kiddin9/Kwrt #1914: 23.05.5分支；6.6.56 内核自定义固件发现luci-app-mosdns无法正常开启

sniper3209, 2024-10-30:
我的新系统是 23.05.5分支；6.6.56 内核:，遇到的问题就是luci-app-mosdns无法启动。即使是卸载包含 mosdns 字符的所有软件包
、删除数据 rm -rf /etc/config/mosdns /etc/mosdns
后重装也不行，又刷回之前用的老版本后一切正常！
所以，大家遇到这个问题了吗？如果解决的？盼回复指导！感谢

xibeilang0616, 2024-10-30:
### **我的版本跟你一样，我的正常。你可能是插件和固件一起编译的吧？如果定制时出了问题就不好解决了。**

### **我的定制固件时是最大化精简定制，刷好后通过软件包里进行所需的插件安装，如果插件出了问题，出厂重置，初始状态，然后再装插件。**

### **缺点是安装大一点的插件比较缓慢，或提示失败，如何提示失败，别动别管他，失败后固件会自动第二次尝试安装，第二遍一般会安装成功。**

sniper3209, 2024-10-30:
> ### **我的版本跟你一样，我的正常。你可能是插件和固件一起编译的吧？如果定制时出了问题就不好解决了，先天性疾病，怎么折腾也没用，因为固件和插件一起定制的根本就没办法卸载，卸载后插件依然在系统中，卸载后再安装，新老插件同时在系统中，磁盘剩余空间会越来越小，直到爆满。**
> ### **我的定制固件时是最大化精简定制，刷好后通过软件包里进行所需的插件安装，如果插件出了问题，卸载重装插件，插件会卸载干净，但是插件的配置文件依然在，等插件重装后，你会发现，插件的设置跟以前一个样。最干净的办法出厂重置，初始状态，然后再装插件。**
> ### **缺点是安装大一点的插件比较缓慢，或提示失败，如何提示失败，别动别管他，失败后固件会自动第二次尝试安装，大概得花十分钟左右的时间，第二遍一般会安装成功。**

感谢你的指导