kiddin9/Kwrt #1929: opkg update 贵站套cf导致全部下载失败

alexallen1, 2024-11-15:
```
[root@Kwrt:01:39 PM ~] # opkg update
Downloading https://dl.openwrt.ai/23.05/targets/mediatek/filogic/6.6.48/Packages.gz
*** Failed to download the package list from https://dl.openwrt.ai/23.05/targets/mediatek/filogic/6.6.48/Packages.gz

Downloading https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/base/Packages.gz
*** Failed to download the package list from https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/base/Packages.gz

Downloading https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/packages/Packages.gz
*** Failed to download the package list from https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/packages/Packages.gz

Downloading https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/luci/Packages.gz
*** Failed to download the package list from https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/luci/Packages.gz

Downloading https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/routing/Packages.gz
*** Failed to download the package list from https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/routing/Packages.gz

Downloading https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9/Packages.gz
*** Failed to download the package list from https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9/Packages.gz

Collected errors:
 * opkg_download: 下载失败 https://dl.openwrt.ai/23.05/targets/mediatek/filogic/6.6.48/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/base/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/packages/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/luci/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/routing/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://dl.openwrt.ai/23.05/packages/aarch64_cortex-a53/kiddin9/Packages.gz, wget returned 8.
 * 
```

MINI-Qiang, 2024-11-26:
kwrt_core 不知道弄啥

alexallen1, 2024-11-27:
晕死啊
https://dl.openwrt.ai/23.05/ 地址直接抹去了吗，影响成千上万用户

ReleasesIT, 2024-11-27:
手动访问了一下是404，看样子是直接移除了旧的软件仓库。这个分支可以放弃了，更新策略这么离谱

laoniuyz, 2024-12-17:
现在这个地址了，https://dl.openwrt.ai/packages-23.05/aarch64_cortex-a53/base

laoniuyz, 2024-12-17:
> 换地址了。 src/gz kwrt_core https://dl.openwrt.ai/23.05/targets/rockchip/armv8/6.6.54 这个没找到
> 
> 下面是更新后的：
> 
> src/gz kwrt_base https://dl.openwrt.ai/packages-24.10/aarch64_generic/base/ src/gz kwrt_packages https://dl.openwrt.ai/packages-24.10/aarch64_generic/packages/ src/gz kwrt_luci https://dl.openwrt.ai/packages-24.10/aarch64_generic/luci src/gz kwrt_routing https://dl.openwrt.ai/packages-24.10/aarch64_generic/routing/ src/gz kwrt_kiddin9 https://dl.openwrt.ai/packages-24.10/aarch64_generic/kiddin9/
> 
> aarch64_generic 换成你自己的平台

23.05不能用24.10，更新会出现问题

AnChiehChen, 2025-03-02:
23.05的软件源现在能正常用了吗